cjcliffe/CubicSDR #364: Feature request: Analog bandwidth selection for BladeRF

piranha32, 2016-06-06:
BladeRF offers selection of bandwidth of the analog frontend filter. It's not clear how the bandwidth is currently selected, but it is too wide: strong signals from one of the spectrum are mirrored on the other end. Please add an option to select the bandwith manually.

piranha32, 2016-06-06:
Problem is more severe than I thought. The screenshot shows images of FM stations leaking to air band.
[image]

cjcliffe, 2016-07-22:
@piranha32 I don't have a complete bandwidth selection solution yet; but can you try this bladeRF setBandwidth patched build to see if it solves the issue -- if so I'll incorporate the patch into the release builds for now.

[CubicSDR-0.2.0-Darwin-rc4-bladerf_patch.dmg.zip](https://github.com/cjcliffe/CubicSDR/files/379235/CubicSDR-0.2.0-Darwin-rc4-bladerf_patch.dmg.zip)

closed 2026-11-14 by cjcliffe