cjcliffe/CubicSDR #366: recursive_mutex_lock frequent crashes due to Rebuffer purge() deletions.

cjcliffe, 2016-06-07:
Addition of the mutex locks to the ReferenceCounter is causing all kinds of new mutex crashes; need to figure out how to handle this cleanly or replace it with something better.

When a ThreadQueue sends a reference counter and the Rebuffer purge()'s the source buffer before a downstream thread is finished we get a crash; this happens in several places and adding more mutex locks seems to make it worse..

vsonnier, 2016-06-08:
Sorry to hear of your problems :(. So, I've taken a closer look to IOThread.h, and have some questions here and there about ReBuffer.getBuffer() implementation.
If I understand correctly:
-  Any buffer of refcount > 0 is considered in use,
-  getBuffer() returns an available buffer, i.e either a collectable one ( ref < 0) or if not possible, a new buffer,
-  Everytime a getBuffer() occurs, all the non-selected buffers with <= 0 potentially collectable are aging, by decrementing their refcount,
- In user code, when a buffer is no longer needed, call decRefCount() on it,
-  If a collectable Buffer reaches -GC_LIMIT, i.e it is "too old" and really collected and freed.

And so my remarks (see comments)
``````cpp
  BufferType *getBuffer() {
        std::lock_guard < std::mutex > lock(m_mutex);

        BufferType* buf = nullptr;
        for (outputBuffersI = outputBuffers.begin(); outputBuffersI != outputBuffers.end(); outputBuffersI++) {
            if (buf == nullptr && (*outputBuffersI)->getRefCount() <= 0) { 
                buf = (*outputBuffersI);
                buf->setRefCount(0);//why not 1 , because the thing is still collectable with a refcount of zero, see purge() ???
            } else if ((*outputBuffersI)->getRefCount() <= 0) { //the aging process, if I get it right...
                (*outputBuffersI)->decRefCount();
            }
        }
        
        if (buf != nullptr) { 
            //Why collecting only the last of the list, and and not all possible, in one pass above ??
            if (outputBuffers.back()->getRefCount() < -REBUFFER_GC_LIMIT) {
                BufferType *ref = outputBuffers.back();
                outputBuffers.pop_back();
                delete ref;
            }
            return buf;
        }
        
...
        }

        buf = new BufferType();
        outputBuffers.push_back(buf);  //Should its refecount not be > 0, else it is collectable, same remark as above ???
        
        return buf;
    }
``````

Which leads to the following like:
``````cpp
  BufferType *getBuffer() {
        std::lock_guard < std::mutex > lock(m_mutex);

        BufferType* buf = nullptr;
        outputBuffersI = outputBuffers.begin();
        while (outputBuffersI != outputBuffers.end())
        {
            if (buf == nullptr && (*outputBuffersI)->getRefCount() <= 0) { 
                buf = (*outputBuffersI);
                buf->setRefCount(1);
                 outputBuffersI++;
            } else if ((*outputBuffersI)->getRefCount() <= 0) {
                 BufferType *deletionCandidate =  (*outputBuffersI); 
                  deletionCandidate->decRefCount();
                 // Maybe deletionCandidate is collectable now
                if (deletionCandidate->getRefCount() < -REBUFFER_GC_LIMIT) {
                    outputBuffersI = outputBuffers.erase(outputBuffersI);
                    delete deletionCandidate;
               } else {
                      outputBuffersI++;
                } 
            }
        }
        
        if (buf != nullptr) { 
            return buf;
        }
        
...
        }

        buf = new BufferType();
        buf->setRefCount(1);
        outputBuffers.push_back(buf); 
        
        return buf;
    }
``````
with outputBuffers changed to be <vector> or <forward_list> to support erase() on iteration.

cjcliffe, 2016-06-08:
@vsonnier interesting; you're exactly right with your evaluation of how it works -- I hadn't considered it getting collected when the buffer is being created but I don't think there's any scenario where getBuffer() and purge() would ever be called at the same time since they're only ever used in the same thread.

cjcliffe, 2016-06-08:
@vsonnier the only issue remaining at the moment in #367 is that there's a buffer left in the garbage collector with a usage of 1; somewhere the reference isn't being decremented when it's done.

Certainly no fault of yours -- these threading issues are the primary cause of random crashes and it's definitely time to get them working correctly.

cjcliffe, 2016-06-09:
Hmm, interesting -- I've been assuming it's the last buffer that's been stuck in the queue.. it actually looks like it might be the *first* one created after adding indexes to the log..

```
Garbage collected demodulator instance 98.904
'DemodulatorPreThreadBuffers' pushed garbage..
'DemodulatorThreadBuffers' purging.. total indexes: 3
'DemodulatorThreadBuffers' pushed garbage..
Garbage collected demodulator instance 98.904
Garbage in queue buffer idx #0, 1 usage(s)
Garbage collecting 1 ReBuffer(s)
Garbage in queue buffer idx #0, 1 usage(s)
Garbage in queue buffer idx #0, 1 usage(s)
```

cjcliffe, 2016-06-09:
Hmm, not always the first one; but have yet to see it be the last one..

cjcliffe, 2016-06-09:
Think I've got it -- looks like a case of never really doing the proper cleanup after transitioning everything to use the mixer instead of individual audio thread per modem :)

cjcliffe, 2016-06-09:
All looks good here now; appears stable on OSX and Windows

closed 2016-06-09 by cjcliffe