cjcliffe/CubicSDR #374: CubicSDR will not launch on Mac OS 10.10.5 since version 0.1.27-alpha and all later releases.

g0oan, 2016-06-16:
Hi,

I wonder if someone could check the compiler setting for the Mac OS version of  0.1.27-alpha and all subsequent please.

While the release notes for version 0.1.27-alpha state "OSX now works with 10.9+ (previously 10.10+)," the programme will not launch on Mac OS 10.10.5. The app icon displays the greyed out, programme not compatible with this version of the system software warning. It is the same for all later releases.

The previous CubicSDR release, version v0.1.26-alpha, works just fine so I'm assuming something in the compiler settings changed between these two versions.

Thanks in advance for any help.

Sean.

cjcliffe, 2016-06-17:
Hmm, interesting I'm running on 10.11.5 here and have had reports of 10.9 working OK so I'm not sure where the disconnect is.. I'll tinker with the build flags and post some test builds here soon.

cjcliffe, 2016-07-14:
@g0oan I've done a full rebuild and checked my 10.9 library linkages for the 0.2.0 rc4 build; let me know if you're still having problems with it:

https://github.com/cjcliffe/CubicSDR/releases/tag/0.2.0-beta-rc4

g0oan, 2016-07-15:
Thank you for that, all working just fine. 

Have tested on Mac OS 10.10 (Yosemite) and 10.11 (El Capitan) on my MacBook Pro and
everything is well.

Thank you once again for the help.

Kind regards,

Sean.

cjcliffe, 2016-07-15:
@g0oan good to hear; I'll close this issue up -- cheers!

closed 2016-07-15 by cjcliffe