cjcliffe/CubicSDR #375: Segfault with hamlib

pwarren, 2016-06-16:
Finally got CAT control working for my OpenRadio SDR: https://github.com/pwarren/open_radio_miniconf_2015/tree/master/openradio_cat

When trying to get CubicSDR to talk to it via the SDR selection dialog I get a segfault, here's a gdb session with a stacktrace, it's repeatable so can always do another if you need more info or different gdb commands, I'm not overly familiar with its use!

```
pwarren@hollis:~/Projects/CubicSDR/build (master)$ gdb ./x64/CubicSDR 
GNU gdb (Debian 7.10-1.1) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./x64/CubicSDR...done.
(gdb) run
Starting program: /home/pwarren/Projects/CubicSDR/build/x64/CubicSDR 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Gtk-Message: Failed to load module "canberra-gtk-module"
Loading:: configuration file '/home/pwarren/.CubicSDR/config.xml'
LoadFromFileXML[error loading]: /home/pwarren/.CubicSDR/config.xml
Loaded 214 rig models via hamlib.
[New Thread 0x7fffe7900700 (LWP 3954)]
[New Thread 0x7fffe70ff700 (LWP 3955)]
[New Thread 0x7fffe68fe700 (LWP 3956)]
[New Thread 0x7fffcea14700 (LWP 3957)]

Audio Device #0 PulseAudio
	Default Output? Yes
	Default Input? Yes
	Input channels: 2
	Output channels: 2
	Duplex channels: 2
	Native formats:
		16-bit signed integer.
		32-bit signed integer.
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
		8000hz
		16000hz
		22050hz
		32000hz
		44100hz
		48000hz
		96000hz

Available vertical sync SwapInterval functions: 
	glxSwapIntervalEXT: No
	DRI2SwapInterval: No
	glxSwapIntervalMESA: No
	glxSwapIntervalSGI: No
No vertical sync swap interval functions available.
Loaded font 'Bitstream Vera Sans Mono' from '/home/pwarren/Projects/CubicSDR/build/x64/vera_sans_mono12.fnt', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/home/pwarren/Projects/CubicSDR/build/x64/vera_sans_mono24.fnt', parsed 255 characters.
[New Thread 0x7fffccdb5700 (LWP 3958)]
SDR enumerator starting.
SoapySDR init..
	API Version: v0.5.0-gb03c4840
	ABI Version: v0.5-2
	Install root: /usr/local
	Loading modules... 
	Available factories...audio, bladerf, null, rtlsdr
[New Thread 0x7fffc757e700 (LWP 3959)]
[Thread 0x7fffc757e700 (LWP 3959) exited]
[New Thread 0x7fffc757e700 (LWP 3960)]
  default_input = True
  default_output = True
  device_id = 0
  driver = audio
[Thread 0x7fffc757e700 (LWP 3960) exited]
  label = PulseAudio
Make device 0
  device_id=0
  origin=https://github.com/pothosware/SoapyAudio

Reporting enumeration complete.
SDR enumerator done.
[Thread 0x7fffccdb5700 (LWP 3958) exited]

Program received signal SIGSEGV, Segmentation fault.
__memcpy_sse2_unaligned ()
    at ../sysdeps/x86_64/multiarch/memcpy-sse2-unaligned.S:33
33	../sysdeps/x86_64/multiarch/memcpy-sse2-unaligned.S: No such file or directory.
(gdb) bt
#0  __memcpy_sse2_unaligned ()
    at ../sysdeps/x86_64/multiarch/memcpy-sse2-unaligned.S:33
#1  0x00007ffff5014c93 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_assign(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#2  0x00007ffff5014fb9 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::operator=(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00000000005927bb in SDRDevicesDialog::OnUseSelected (this=0x38c7090, 
    event=...)
    at /home/pwarren/Projects/CubicSDR/src/forms/SDRDevices/SDRDevices.cpp:284
#4  0x00007ffff605559e in wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#5  0x00007ffff61dba87 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#6  0x00007ffff61dbe6e in wxEvtHandler::SearchDynamicEventTable(wxEvent&) ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#7  0x00007ffff61dbeff in wxEvtHandler::TryHereOnly(wxEvent&) ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#8  0x00007ffff61dbfb3 in wxEvtHandler::ProcessEventLocally(wxEvent&) ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#9  0x00007ffff61dc015 in wxEvtHandler::ProcessEvent(wxEvent&) ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#10 0x00007ffff61dbd87 in wxEvtHandler::SafelyProcessEvent(wxEvent&) ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#11 0x00007ffff6e10cbb in ?? ()
   from /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#12 0x00007ffff38fc99c in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#13 0x00007ffff32d1fa5 in g_closure_invoke ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#14 0x00007ffff32e3fc1 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#15 0x00007ffff32ec7f9 in g_signal_emit_valist ()
---Type <return> to continue, or q <return> to quit---
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#16 0x00007ffff32ed08f in g_signal_emit ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#17 0x00007ffff3a1431c in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#18 0x00007ffff38fb134 in gtk_propagate_event ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#19 0x00007ffff38fb4eb in gtk_main_do_event ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#20 0x00007ffff356fc9c in ?? ()
   from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#21 0x00007ffff23291a7 in g_main_context_dispatch ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#22 0x00007ffff2329400 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#23 0x00007ffff2329722 in g_main_loop_run ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#24 0x00007ffff38fa567 in gtk_main ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#25 0x00007ffff6de29c5 in wxGUIEventLoop::DoRun() ()
   from /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#26 0x00007ffff60976f3 in wxEventLoopBase::Run() ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#27 0x00007ffff605ca06 in wxAppConsoleBase::MainLoop() ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#28 0x00007ffff60e9310 in wxEntry(int&, wchar_t**) ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#29 0x00000000004a2b1c in main (argc=1, argv=0x7fffffffe268)
    at /home/pwarren/Projects/CubicSDR/src/CubicSDR.cpp:25

```

cjcliffe, 2016-06-17:
Thanks @pwarren -- there's been word of crashes with CubicSDR and hamlib and this will help for sure.

Cheers!

pwarren, 2016-06-24:
Ok, I think I've found the source of the issue, it might be SoapyAudio rather than CubicSDR.

My steps to reproduce:

`cd SoapyAudio/build
rm -rf *
cmake -DUSE_HAMLIB ../
make && sudo make install && sudo ldconfig`

`cd ../../CubicSDR/build
rm -rf *
cmake -DUSE_HAMLIB ../ && make
./x64/CubicSDR
`
then select PulseAudio for the SDR choices thing, and set the CAT parameters there, click start: SEGFAULT.

This also happens when using CubicSDR build with a plain `cmake ../ && make

cjcliffe, 2016-06-25:
Hmm, have you pulled the latest sources for both SoapyAudio and CubicSDR? 

CubicSDR's internal version of RtAudio was updated recently; and if it doesn't match SoapyAudio's version there was an immediate crash on OSX, Linux and Windows when it loaded the module.

closed 2016-08-16 by cjcliffe