cjcliffe/CubicSDR #377: Gdr Error 'BadValue'

N4CVX, 2016-06-23:
dave@TheMightyWurlitzer-II:~/CubicSDR/build/x64$ ./CubicSDR
Loading:: configuration file '/home/dave/.CubicSDR/config.xml'

(CubicSDR:30090): Gdk-ERROR **: The program 'CubicSDR' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 254 error_code 2 request_code 155 (GLX) minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
Trace/breakpoint trap (core dumped)
dave@TheMightyWurlitzer-II:~/CubicSDR/build/x64$ ./CubicSDR -GDK_SYNCHRONIZE
Unknown option 'GDK_SYNCHRONIZE'
Usage: CubicSDR [-h] [-c <str>] [-m <str>]
  -h, --help         	Command line parameter help
  -c, --config=<str> 	Specify a named configuration to use, i.e. '-c ham'
  -m, --modpath=<str>	Load modules from suppplied path, i.e. '-m ~/SoapyMods/'
dave@TheMightyWurlitzer-II:~/CubicSDR/build/x64$ GDK_SYNCHRONIZE
GDK_SYNCHRONIZE: command not found
dave@TheMightyWurlitzer-II:~/CubicSDR/build/x64$

## Soapy SDR -- the SDR abstraction library
######################################################

Probe device 

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=null
  hardware=null

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 0 Rx, 0 Tx
  Timestamps: NO

*** Error in `SoapySDRUtil': munmap_chunk(): invalid pointer: 0x0000000001decbe8 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x77725)[0x7f9decf8a725]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x1a8)[0x7f9decf96c18]
/lib/x86_64-linux-gnu/libc.so.6(+0x39fe8)[0x7f9decf4cfe8]
/lib/x86_64-linux-gnu/libc.so.6(+0x3a035)[0x7f9decf4d035]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf7)[0x7f9decf33837]
SoapySDRUtil(_start+0x29)[0x4032f9]
======= Memory map: ========
00400000-00410000 r-xp 00000000 08:01 39321607                           /usr/local/bin/SoapySDRUtil
0060f000-00610000 r--p 0000f000 08:01 39321607                           /usr/local/bin/SoapySDRUtil
00610000-00611000 rw-p 00010000 08:01 39321607                           /usr/local/bin/SoapySDRUtil
01dd7000-01e45000 rw-p 00000000 00:00 0                                  [heap]
7f9de95c2000-7f9de95c3000 ---p 00000000 00:00 0 
7f9de95c3000-7f9de9dc3000 rw-p 00000000 00:00 0 
7f9de9dc3000-7f9de9df3000 r-xp 00000000 08:01 39321608                   /usr/local/lib/libSoapySDR.so.0.4.0
7f9de9df3000-7f9de9ff3000 ---p 00030000 08:01 39321608                   /usr/local/lib/libSoapySDR.so.0.4.0
7f9de9ff3000-7f9de9ff4000 r--p 00030000 08:01 39321608                   /usr/local/lib/libSoapySDR.so.0.4.0
7f9de9ff4000-7f9de9ff5000 rw-p 00031000 08:01 39321608                   /usr/local/lib/libSoapySDR.so.0.4.0
7f9de9ffb000-7f9dea00e000 r-xp 00000000 08:01 39321603                   /usr/local/lib/libmirsdrapi-rsp-x86_64-1.8.1.so
7f9dea00e000-7f9dea20d000 ---p 00013000 08:01 39321603                   /usr/local/lib/libmirsdrapi-rsp-x86_64-1.8.1.so
7f9dea20d000-7f9dea20e000 r--p 00012000 08:01 39321603                   /usr/local/lib/libmirsdrapi-rsp-x86_64-1.8.1.so
7f9dea20e000-7f9dea210000 rw-p 00013000 08:01 39321603                   /usr/local/lib/libmirsdrapi-rsp-x86_64-1.8.1.so
7f9dea210000-7f9debd73000 rw-p 00000000 00:00 0 
7f9debd73000-7f9debd85000 r-xp 00000000 08:01 40632658                   /usr/local/lib/SoapySDR/modules/libsdrPlaySupport.so
7f9debd85000-7f9debf84000 ---p 00012000 08:01 40632658                   /usr/local/lib/SoapySDR/modules/libsdrPlaySupport.so
7f9debf84000-7f9debf85000 r--p 00011000 08:01 40632658                   /usr/local/lib/SoapySDR/modules/libsdrPlaySupport.so
7f9debf85000-7f9debf86000 rw-p 00012000 08:01 40632658                   /usr/local/lib/SoapySDR/modules/libsdrPlaySupport.so
7f9debf8b000-7f9debf92000 r-xp 00000000 08:01 40636919                   /lib/x86_64-linux-gnu/librt-2.23.so
7f9debf92000-7f9dec191000 ---p 00007000 08:01 40636919                   /lib/x86_64-linux-gnu/librt-2.23.so
[... 34 lines cut ...]
7f9ded4f8000-7f9ded4f9000 rw-p 00015000 08:01 40636803                   /lib/x86_64-linux-gnu/libgcc_s.so.1
7f9ded4fb000-7f9ded66d000 r-xp 00000000 08:01 38799358                   /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
7f9ded66d000-7f9ded86d000 ---p 00172000 08:01 38799358                   /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
7f9ded86d000-7f9ded877000 r--p 00172000 08:01 38799358                   /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
7f9ded877000-7f9ded879000 rw-p 0017c000 08:01 38799358                   /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
7f9ded879000-7f9ded87d000 rw-p 00000000 00:00 0 
7f9ded883000-7f9ded8ba000 r-xp 00000000 08:01 39329536                   /usr/local/lib/libSoapySDR.so.0.5.0
7f9ded8ba000-7f9dedaba000 ---p 00037000 08:01 39329536                   /usr/local/lib/libSoapySDR.so.0.5.0
7f9dedaba000-7f9dedabb000 r--p 00037000 08:01 39329536                   /usr/local/lib/libSoapySDR.so.0.5.0
7f9dedabb000-7f9dedabc000 rw-p 00038000 08:01 39329536                   /usr/local/lib/libSoapySDR.so.0.5.0
7f9dedac3000-7f9dedae9000 r-xp 00000000 08:01 40636737                   /lib/x86_64-linux-gnu/ld-2.23.so
7f9dedc7b000-7f9dedc99000 r-xp 00000000 08:01 40632358                   /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f9dedc99000-7f9dedc9a000 r--p 0001d000 08:01 40632358                   /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f9dedc9a000-7f9dedc9b000 rw-p 0001e000 08:01 40632358                   /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f9dedcc1000-7f9dedcc3000 rw-p 00000000 00:00 0 
7f9dedce6000-7f9dedce8000 rw-p 00000000 00:00 0 
7f9dedce8000-7f9dedce9000 r--p 00025000 08:01 40636737                   /lib/x86_64-linux-gnu/ld-2.23.so
7f9dedce9000-7f9dedcea000 rw-p 00026000 08:01 40636737                   /lib/x86_64-linux-gnu/ld-2.23.so
7f9dedcea000-7f9dedcec000 rw-p 00000000 00:00 0 
7f9dedcec000-7f9dedcf2000 rw-p 00000000 00:00 0 
7ffcca0bc000-7ffcca0dd000 rw-p 00000000 00:00 0                          [stack]
7ffcca1b3000-7ffcca1b5000 r--p 00000000 00:00 0                          [vvar]
7ffcca1b5000-7ffcca1b7000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Aborted (core dumped)

cjcliffe, 2016-06-23:
Thanks for the report; I think I'm going to try FlatPak as well as AppImageKit and keep them both up for awhile for testing -- I'll post back here when there's a new build to try.

cjcliffe, 2016-07-14:
@N4CVX let me know if you're still seeing this error with latest pushes to master -- I'd also be interested to know if the builds at https://github.com/cjcliffe/CubicSDR/releases/tag/0.2.0-beta-rc4 work for you or are showing the same issue.

cjcliffe, 2016-07-28:
I'll close this one up for now; please submit a new issue if you're having problems with the new builds -- thanks!

closed 2016-07-28 by cjcliffe