cjcliffe/CubicSDR #383: Sluggish performance and looking for fonts in wrong location

avsa242, 2016-06-25:
Starting with 46bf72aa0f7f94a4ad8a1f3a97808a61b0250a53, it seems CubicSDR looks for fonts under /usr/local/bin/fonts, rather than /usr/local/share/cubicsdr/fonts. Going by the messages in the terminal, the path it looks for seems to continuously grow recursively as well:
`
Font file /usr/local/bin/fonts/vera_sans_mono12.fnt does not exist?
Font file /usr/local/bin/usr/local/bin/fonts/vera_sans_mono12.fnt does not exist?
Font file /usr/local/bin/usr/local/bin/usr/local/bin/fonts/vera_sans_mono12.fnt does not exist?
`
either that or it's the same path every time, but it just happens to look recursive because of how it is output on the terminal. This causes no fonts to be rendered and the performance to be poor.

vsonnier, 2016-06-25:
@avsa242 Indeed I broke the font load by resource folder. As a temporary fix, copy the 'fonts' dir alongside the CubicSDR executable, it should load fine. 

Hopefully fixed by #385 for good.

cjcliffe, 2016-06-29:
@avsa242 latest commit to master will now install fonts to /usr/local/share/cubicsdr/fonts/ where it's expected so that the font loader doesn't try to look in /usr/local/bin/fonts/.

closed 2016-06-29 by cjcliffe