cjcliffe/CubicSDR #387: active demodulator frequency

imesg, 2016-06-27:
I can now select a demodulator frequency only in 1mhz resolution in the lower waterfall (example: I want to select 134.775 mhz I can select either 134.000 or 135.000, nothing in between).

Is there a setting for this?

Thanks

cjcliffe, 2016-06-27:
@imesg there should be a red box around the 1Mhz digit in the frequency selector; that indicates it's set as the 'snap' value.   

Right-clicking another digit will change the snap value to that digit, just right-click the same digit again to turn it off.

Some additional information about tuning can now also be found here:

http://cubicsdr.readthedocs.io/en/latest/application-window.html#tuning-bar

imesg, 2016-06-27:
Should've known it would be something easy.

cjcliffe, 2016-06-27:
@imesg not a problem, closing this issue up; cheers!

closed 2016-06-27 by cjcliffe