cjcliffe/CubicSDR #388: I/Q input via generic audio device

tonyc, 2016-06-27:
Hi!

I'd love to be able to feed the I/Q output from my KX3 into Cubic. I would love to see support for I/Q over generic sound devices to allow this.

cjcliffe, 2016-06-27:
@tonyc If you check out the SoapyAudio (https://github.com/pothosware/SoapyAudio) module you'll find support for Mono L/R, I/Q, Q/I, and sample offsets.  

There's also optional Hamlib support (-DUSE_HAMLIB=1 cmake option) for SoapyAudio which looks like it supports the KX3 via a quick check with rigctld:
```
ccliffe$ rigctld --list | grep KX3
  229  Elecraft               K3/KX3                  20130118        Beta
ccliffe$ 
```

Let me know if you have any questions on getting it running.

tonyc, 2016-06-27:
@cjcliffe That's precisely what I was looking for! Thanks!

tonyc, 2016-06-28:
@cjcliffe I've got it working pretty well, although I can't seem to get serial control working with the KX3. I've got the rate/etc all set correctly, and my KX3 appears as either /dev/cu.usbserial-A703D7SH or /dev/tty.usbserial-A703D7SH -- setting either as the "Rig Serial Port" in the setup window doesn't seem to have any effect on the UI keeping track of the VFO/etc. 

Any hints, or places I could check to make sure Cubic is speaking to the radio?

cjcliffe, 2016-06-28:
Hmm; have you tried using the Rig Control in SoapyAudio itself?  When enabled you choose the Rig, Serial and Rate from the SDR Devices Dialog instead of the stuff on the Rig Menu (they're not quite the same).

The SoapyAudio built-in one will just listen to center frequency change requests from CubicSDR; the rig control features don't affect it.

Alternately if you're already doing that; try clearing the rig settings for SoapyAudio in SDR Devices Dialog and only use the CubicSDR Rig Menu settings.

tonyc, 2016-06-28:
I've tried configuring the port in the SoapyAudio dialog that shows up when I first start CubicSDR - the settings in the "Rig" menu then seem to mirror what I choose, so I thought they were the same. 

Perhaps I'm missing a plug-in? I'm running CubicSDR 0.1.20 for OSX, and the only thing in the menu that resembles rig control is Settings -> Rig Control, Rig Serial Rate, and Rig Serial Port. 

Incidentally, there are no other options under "Configure -> Rig Serial Port", nor does a dialog appear when I choose it.

Would you prefer I open another ticket for this? :)

tonyc, 2016-06-28:
As an update, playing with the "Frequency" digits causes my KX3 to actually change frequency, but spinning the radio VFO does not update the frequency displayed in CubicSDR -- so it at least seems to be working one way, but CubicSDR does not seem to be polling for the current VFO frequency. I'm not sure if this is by designed or just hasn't been implemented yet :)

cjcliffe, 2016-06-28:
@tonyc I'd try 0.2.0-rc2 build from https://github.com/cjcliffe/CubicSDR/releases with built-in SoapyAudio and Rig Control -- SoapyAudio module doesn't have Rig Control enabled since the regular Rig controls + SoapyAudio should now cover the same use cases -- Many of the rig features have changed/tweaked since 0.1.20 and it might work better now.

tonyc, 2016-06-28:
Perfect! That resolved everything! Thanks! 😎

closed 2016-06-28 by tonyc

cjcliffe, 2016-06-28:
Lol; glad that did it -- cheers!