cjcliffe/CubicSDR #393: Compile Issue

heath33, 2016-06-30:
Compile failing at:

`  _/usr/local/src/CubicSDR/src/modules/modem/analog/ModemFM.h:19:5: error: ‘freqdem’ does not name a type
     freqdem demodFM;
     ^_ `

I saw the other post about the different versions of liquid-dsp.  I can install liquid-dsp from the tarball, but it will not install from the github source (as per the instructions on CubicSDR.)  So I have liquid-dsp installed, but CubicSDR fails, I can't install liquid-dsp per guidance, it fails here

`In file included from src/dotprod/src/dotprod_cccf.mmx.c:49:0:
src/dotprod/src/dotprod_cccf.mmx.c: In function ‘dotprod_cccf_execute_mmx’:
/usr/lib/gcc/i686-linux-gnu/5/include/pmmintrin.h:50:1: error: inlining failed in call to always_inline ‘_mm_addsub_ps’: target specific option mismatch
 _mm_addsub_ps (__m128 __X, __m128 __Y)`

The system I am trying to install this on should be up to date with all dependencies met.
System:    Host: Gateway Kernel: 4.6.0-1-686-pae i686 (32 bit) Desktop: Cinnamon 3.0.4
           Distro: Debian GNU/Linux stretch/sid
Machine:   System: Gateway (portable) product: MT6705 v: 3408385R
           Mobo: Gateway model: N/A v: 77.10 Bios: Phoenix v: 77.12 date: 04/18/2007
Battery    BAT1: charge: 27.4 Wh 98.4% condition: 27.9/48.0 Wh (58%)
CPU:       Dual core Intel T2060 (-MCP-) cache: 1024 KB 
           clock speeds: max: 1600 MHz 1: 1333 MHz 2: 1067 MHz
Graphics:  Card: Intel Mobile 945GM/GMS 943/940GML Express Integrated Graphics Controller
           Display Server: X.Org 1.18.3 drivers: intel (unloaded: fbdev,vesa) Resolution: 1280x800@60.00hz
           GLX Renderer: Mesa DRI Intel 945GM x86/MMX/SSE2 GLX Version: 2.1 Mesa 11.2.2

cjcliffe, 2016-06-30:
@heath33 Ahoy -- You'll need a newer liquid-dsp somehow; older versions of liquid-dsp had an integrated FM modem but there's now a separate modulator and demodulator.  There's other compile errors past that you'll get with an older version as well.

I'd post your liquid-dsp compile error over at the liquid-dsp issues https://github.com/jgaeddert/liquid-dsp/issues but I'll leave it open here as well in case myself or someone here finds a solution first.

heath33, 2016-07-01:
I've posted the issue on liquid-dsp's issues page at https://github.com/jgaeddert/liquid-dsp/issues/47

I get the same compile break on two different machines following the instructions on the CubicSDR linux build page.  One is a fresh Debian/Jessie install on an AMD i386 architecture and one is on a Debian/Sid Intel i386 machine.  On both, pulling the source code from liquid-dsp get http://liquidsdr.org/downloads/liquid-dsp-1.2.0.tar.gz results in successful build, however CubicSDR does not build as described above.

vsonnier, 2017-11-01:
The problem has been solved in current libliquid v1.31 and it builds cleanly now on Ubuntun, Mint, and others, so I'm closing this one.

closed 2017-11-01 by vsonnier