cjcliffe/CubicSDR #399: Bandwidth selection

frspin, 2016-07-06:
In graphic window now is possible to modify bandwidth using mouse over bandwidth limit.
This modify bandwidth (low and hight values) for FM and AM modes. Only low for LSB and only hight for USB. In this modes the other value is fix to 0.0
If possible, make this "zero value" modifiable as the other, for getting very narrow filter without modify centre frequency.

Regards
Franco Spinelli
IW2DHW

cjcliffe, 2016-07-07:
@frspin thanks for the suggestion; I will try to implement it in a future version -- I'll have to think about how to make it both interactive and a toggle (like ALT for PPM) for the tuning bar frequency.

frspin, 2016-07-07:
Thanks.
On graphic window (the upper window, with detail of tuned freq.) you can
use the same cursor change as now for lower/upper frequency.
On tuning bar you can use two number for bandwidth. A "lower" and a "upper"
value with 0 for lower and actual value for upper (in LSB this values need
to be negative for visual purpose only)

Regards

Franco Spinelli
IW2DHW

vpnels, 2016-12-12:
I second the request for ability to move center frequency within bandwidth. This is important for CW operators. We would like to be able to put 150 Hz filter around a signal 1000 Hz from 0 frequency.

Pascal -ac7n