cjcliffe/CubicSDR #402: Bias Tee Settings

darksidelemm, 2016-07-11:
I use an antenna with a mast-head low-noise preamp. While my Airspy has a bias-tee option that works fine under applications like SDR-Console, CubicSDR does not have an option to turn it on!

I guess the same option would be applicable to devices like the HackRF, and maybe the BladeRF?

cjcliffe, 2016-07-11:
The latest Airspy module is an experimental version I've developed since the last release -- I'll have the Bias Tee option available before the final 0.2.0 build is ready; you can find the specific SoapyAirspy issue here:  https://github.com/pothosware/SoapyAirspy/issues/4

It's a fairly simple addition I'll see if I can get it implemented tonight so it's ready for the next build and leave this open to remind myself.

cjcliffe, 2016-07-12:
Bias Tee option has been added to SoapyAirspy module and will be included in the CubicSDR 0.2.0-rc4 build soon.

Cheers!

closed 2016-07-12 by cjcliffe