cjcliffe/CubicSDR #403: error: no match for ‘operator=’ (operand types are ‘std::vector<fftwf_plan_s*>’ and ‘std::nullptr_t’)

stefantalpalaru, 2016-07-11:
I get this compilation error with fftw-3.3.4 and gcc-5.4.0 :

/usr/bin/x86_64-pc-linux-gnu-g++  -DCUBICSDR_VERSION=\"0.2.0-beta-rc3\" -DRES_FOLDER=\"/usr/share/cubicsdr/\" -DUSE_FFTW3=1 -DUSE_SOAPY_SDR=1 -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D__LINUX_ALSA__ -D__UNIX_JACK__ -D__WXGTK__ -isystem /usr/lib64/wx/include/gtk2-unicode-3.0 -isystem /usr/include/wx-3.0 -I/var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/forms/SDRDevices -I/var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/forms/DigitalConsole -I/var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/sdr -I/var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/demod -I/var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/modules -I/var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/modules/modem -I/var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/modules/modem/digital -I/var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/modules/modem/analog -I/var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/audio -I/var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/util -I/var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/panel -I/var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/visual -I/var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/process -I/var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/ui -I/var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/rig -I/var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src -I/var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/external/rtaudio -I/var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/external/lodepng -I/var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/external/tinyxml -I/var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/external/cubicvr2/math   -DNDEBUG -O3 -march=native -pipe  -pthread -fvisibility-inlines-hidden   -std=c++0x -pthread -Wall -Wextra -fvisibility=hidden -std=gnu++11 -o CMakeFiles/CubicSDR.dir/src/process/ScopeVisualProcessor.cpp.o -c /var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/process/ScopeVisualProcessor.cpp
/var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/process/ScopeVisualProcessor.cpp: In constructor ‘ScopeVisualProcessor::ScopeVisualProcessor()’:
/var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/process/ScopeVisualProcessor.cpp:13:15: **error: no match for ‘operator=’ (operand types are ‘std::vector<fftwf_plan_s*>’ and ‘std::nullptr_t’)**
     fftw_plan = nullptr;
               ^
In file included from /usr/lib/gcc/x86_64-pc-linux-gnu/5.4.0/include/g++-v5/vector:69:0,
                 from /usr/lib/gcc/x86_64-pc-linux-gnu/5.4.0/include/g++-v5/queue:61,
                 from /var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/util/ThreadQueue.h:11,
                 from /var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/process/VisualProcessor.h:4,
                 from /var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/process/ScopeVisualProcessor.h:3,
                 from /var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/process/ScopeVisualProcessor.cpp:1:
/usr/lib/gcc/x86_64-pc-linux-gnu/5.4.0/include/g++-v5/bits/vector.tcc:167:5: note: candidate: std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(const std::vector<_Tp, _Alloc>&) [with _Tp = fftwf_plan_s*; _Alloc = std::allocator<fftwf_plan_s*>]
     vector<_Tp, _Alloc>::
     ^
/usr/lib/gcc/x86_64-pc-linux-gnu/5.4.0/include/g++-v5/bits/vector.tcc:167:5: note:   no known conversion for argument 1 from ‘std::nullptr_t’ to ‘const std::vector<fftwf_plan_s*>&’
In file included from /usr/lib/gcc/x86_64-pc-linux-gnu/5.4.0/include/g++-v5/vector:64:0,
                 from /usr/lib/gcc/x86_64-pc-linux-gnu/5.4.0/include/g++-v5/queue:61,
                 from /var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/util/ThreadQueue.h:11,
                 from /var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/process/VisualProcessor.h:4,
                 from /var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/process/ScopeVisualProcessor.h:3,
                 from /var/tmp/portage/net-wireless/cubicsdr-0.2.0_rc3/work/cubicsdr-0.2.0_rc3/src/process/ScopeVisualProcessor.cpp:1:
/usr/lib/gcc/x86_64-pc-linux-gnu/5.4.0/include/g++-v5/bits/stl_vector.h:448:7: note: candidate: std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(std::vector<_Tp, _Alloc>&&) [with _Tp = fftwf_plan_s*; _Alloc = std::allocator<fftwf_plan_s*>]
       operator=(vector&& __x) noexcept(_Alloc_traits::_S_nothrow_move())
       ^
/usr/lib/gcc/x86_64-pc-linux-gnu/5.4.0/include/g++-v5/bits/stl_vector.h:448:7: note:   no known conversion for argument 1 from ‘std::nullptr_t’ to ‘std::vector<fftwf_plan_s*>&&’
/usr/lib/gcc/x86_64-pc-linux-gnu/5.4.0/include/g++-v5/bits/stl_vector.h:470:7: note: candidate: std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(std::initializer_list<_Tp>) [with _Tp = fftwf_plan_s*; _Alloc = std::allocator<fftwf_plan_s*>]
       operator=(initializer_list<value_type> __l)
       ^
/usr/lib/gcc/x86_64-pc-linux-gnu/5.4.0/include/g++-v5/bits/stl_vector.h:470:7: note:   no known conversion for argument 1 from ‘std::nullptr_t’ to ‘std::initializer_list<fftwf_plan_s*>’

cjcliffe, 2016-07-11:
@stefantalpalaru thanks will fix this up -- I've likely broken the fftw support recently; haven't been testing with it here since it's no longer required and is slower compared to liquid-dsp internal FFT.

cjcliffe, 2016-07-20:
@stefantalpalaru since liquid-dsp can use FFTW3 and will wrap the calls I make to liquid-dsp   anyways I've decided to just remove the redundant support in CubicSDR to solve this.

Though I'd note it's not actually recommended to build liquid-dsp with FFTW3 since it's about ~5-10% slower when enabled and tends to use more memory.

Closing this one up; cheers!

closed 2016-07-20 by cjcliffe