cjcliffe/CubicSDR #404: shortcut key

yo2ldk, 2016-07-13:
Hi,

 as I see, in this not work shortcut key "F" for fm, C for cw (that mode not appear in right, beside others), D for dsb etc...
 can you re-enable?

  thank you in advance !

cjcliffe, 2016-07-28:
@yo2ldk Currently there's only:

- 'A' for AM
- 'F' for FM (which you can press more than once to toggle between FM modes)
- 'U' for USB
- 'L' for LSB

'D' is currently reserved for "delete" but I'm considering changing that to DEL/Backspace

You can find the current work-in-progress manual for some more info at: 
http://cubicsdr.readthedocs.io 

Cheers!

yo2ldk, 2016-07-29:
Ok, Thank you for reply and also for your hard work to do something so nice and useful for us !
   my best regards !

Toontje, 2016-12-31:
Yes, i vote for the C as well.

cjcliffe, 2021-03-20:
Linking to #341  -- may move to shift+ modifier for all modulation type selections