cjcliffe/CubicSDR #409: Memory allocation error for large bandwidth/sample rate

mehdideveloper, 2016-07-24:
I've compiled CubicSDR 0.2.0-beta-rc2 from source.
When I try to set sample rate to more than 10MHZ, it crashes and gives this error:

> [ERROR] bladeRF: Invalid sampling mode '''
[ERROR] bladeRF: Invalid loopback setting ''
CubicSDR(4148,0x7fff74059000) malloc: *** mach_vm_map(size=7575166089418604544) failed (error code=3)
*** error: can't allocate region
*** set a breakpoint in malloc_error_break to debug
Terminating SDR thread..

Then I updated it to CubicSDR 0.2.0-beta-rc4 and now I get random errors.
For example this erros is returned for sample rate > 10MHZ.

> CubicSDR(6056,0x7fff74059000) malloc: *** mach_vm_map(size=562992903294976) failed (error code=3)
*** error: can't allocate region
*** set a breakpoint in malloc_error_break to debug
Terminating SDR thread..
ERROR: thread '9SDRThread' has not terminated in time ! (> 3000 ms)

And this error is returned for sample rate 2MHZ:

> Segmentation fault: 11

And then sometimes it works flawlessly with 20MHZ sample rate!

Here's my Soapy* info:
API Version: v0.5.0-g37c7ea4b
ABI Version: v0.5-2
Available factories...audio, bladerf, null, remote, rtlsdr, sdrplay,

MacBook Pro 13" 2015 (8GB RAM, about 5GB used at the time of test)
Any ideas?

cjcliffe, 2016-07-24:
@mehdideveloper Hmm -- seems like an uninitialized variable being used for a buffer size -- what's the last time you've updated SoapySDR, SoapyAudio and the other modules?  There was a recent problem with them mismatching that caused RtAudio to crash things.

Are you able to build with XCode and set an 'all exceptions' breakpoint and run from there?  That might give a better indication of the crash point.

cjcliffe, 2016-07-24:
@mehdideveloper a quick scan of the SoapyBladeRF code it looks like the MTU is 0 until the stream is initialized but that seemed like the most likely source -- so I should probably handle that better  somewhere but otherwise I can't immediately see why it's attempting to allocate sizes like 7575166089418604544 and 562992903294976.

cjcliffe, 2016-07-24:
@mehdideveloper What does it report for "Stream MTU: " in the log?   I'm thinking setupStream failed and I've failed to handle it properly.  Going to patch a build so that perhaps it won't crash and give us some information instead.

cjcliffe, 2016-07-24:
Here's a patched build from the bladerf_patch branch I've been working on -- I can't reproduce the issue here with HackRF unless I break the stream setup on purpose -- but that should be handled more cleanly now and you can re-attempt to start the device if it doesn't start and we'll just see the init failure in the log again.

[CubicSDR-0.2.0-Darwin-rc4-bladerf_patch2.dmg.zip](https://github.com/cjcliffe/CubicSDR/files/380352/CubicSDR-0.2.0-Darwin-rc4-bladerf_patch2.dmg.zip)

mehdideveloper, 2016-07-24:
@cjcliffe This patch works flawlessly. I tried 2MHZ, 10MHZ, 20MHZ, and 26MHZ.
Every time it worked without any problems and it does not fail anymore.
Are there more tests I can do to help you troubleshoot it better?

cjcliffe, 2016-07-24:
@mehdideveloper that's great to hear -- I can't think of anything specific to try for now but feel free to open another issue if you find anything.  I've merged the fixes to master branch as well.

Cheers!

closed 2016-07-24 by cjcliffe