cjcliffe/CubicSDR #411: 'No SoapySDR module found' error with custom built Windows builds

vsonnier, 2016-07-26:
@cjcliffe Hello ! I'm trying to compile the master for Win64, reusing all the other v0.20-rc4-win64 release binaries, which is what I do to use/test the master usually. What I'm getting now is this:
[image]
poping up like mad.
 
The CubicSDR exe from https://github.com/cjcliffe/CubicSDR/releases/tag/0.2.0-beta-rc4 actually runs fine, though. With some build specific settings maybe ?

A workaround is to provide the module path on command line explicitly like `...\CubicSDR.exe -m ./modules/`

I actually started to observe this when attempting to compile the SoapySDR v0.5+ some time ago, but since it is built-in now the problem shows up all the time.

cjcliffe, 2016-07-26:
Hmm, for local testing in Windows I use the active SoapySDR paths -- Just make sure the soapy module install prefixes are set to C:\Program Files\SoapySDR\ and the SoapySDR bin folder is in your path as well as the SoapySDR environment variable should be C:\Program Files\SoapySDR\ (this stuff changes if you installed pothos)  

From there you copy any relevant bins for the modules into the SoapySDR bins folder.

guruofquality, 2016-07-26:
> I actually started to observe this when attempting to compile the SoapySDR v0.5+ some time ago, but since it is built-in now I can no longer dodge the problem. 

Perhaps this explains it? So, I havent tagged 0.5.1 yet, but there was a change to the modules directory to address a multi-version install on debian packaging issue... more details: https://github.com/pothosware/SoapySDR/pull/85

I wasn’t sure how this would manifest because I thought cubic used its own loader, but I can add back the original search path. Its just looking in INSTALL_ROOT/lib/SoapySDR/modules.0.5-3 But any recompiled modules should install to that location.

vsonnier, 2016-08-05:
@cjcliffe @guruofquality  OK, mystery solved. I managed to run the https://github.com/cjcliffe/CubicSDR/releases/tag/0.2.0-beta-rc5 executable using the Git Bash on Windows, and to my surprise I actually got std::cout traces! So it turns out `Loading bundled SoapySDR module...` messages showed up, and this could only occur if `BUNDLE_SOAPY_MODS` preprocessor definition is set, as the following code SDREnumerator.cpp (111) shows:
``````cpp
			wxFileName exePath = wxFileName(wxStandardPaths::Get().GetExecutablePath());
			std::vector<std::string> localMods = SoapySDR::listModules(exePath.GetPath().ToStdString() + "/modules/");
			for (std::vector<std::string>::iterator mods_i = localMods.begin(); mods_i != localMods.end(); mods_i++) {
				wxGetApp().sdrEnumThreadNotify(SDREnumerator::SDR_ENUM_MESSAGE, "Initializing bundled SoapySDR module " + (*mods_i) + "..");
				std::cout << "Loading bundled SoapySDR module " << (*mods_i) << ".." << std::endl;
				SoapySDR::loadModule(*mods_i);

		} 
``````
Which clearly loads the modules explicitly from CubicSDR.exe/modules. On the other hand, on my setup I didn't configure CMake  `BUILD_INSTALLER` and then `BUNDLE_SOAPY_MODS` so modules were searched from `SOAPY_INSTALL_ROOT` and subdirs as @guruofquality said, making the executable position dependent, luckily to the release one.

Maybe the CubicSDR Wiki should indicate that the user should set `BUILD_INSTALLER` and `BUNDLE_SOAPY_MODS` to prevent those kinds of path problems.

vsonnier, 2016-08-05:
I leave it open as a Building doc issue for the Wiki :)

vsonnier, 2016-08-14:
@cjcliffe  I've updated the Windows build instructions to take into account the modules related variables. Closing this issue now.

closed 2016-08-14 by vsonnier