cjcliffe/CubicSDR #412: FM broadcast de-emphasis

Alrescha-git, 2016-07-26:
I'd like to suggest adding de-emphasis to the FM and (especially) FMS modes.

While the sound quality is currently excellent, it is a little too harsh to be listening for pleasure.  Adding 50us (EU)/75us (NA) would be a big help.

cjcliffe, 2016-07-26:
@Alrescha-git agreed; this should be fairly straightforward to add to FM Stereo modem now; will take a look before doing the 0.2.0 final build

cjcliffe, 2016-07-28:
@Alrescha-git I've added de-emphasis support for FM Stereo which will lead the way as the first use of the modem properties UI in 0.2.0.

I'll be working on filtering options for the modem properties of other modems soon so I'll leave this issue open as a reminder to implement the de-emphasis for FM Mono.

Alrescha-git, 2016-07-28:
Thank you.  I am looking forward to giving it a try.

cjcliffe, 2016-07-28:
@Alrescha-git will be included in the rc5 build; cheers!

Alrescha-git, 2016-07-29:
Downloaded and gave it a listen - fantastic.

cjcliffe, 2016-07-29:
@Alrescha-git sounding great here as well -- better than expected; thanks for reminding me to finally look into this.  Hoping to add some more filters to other modes with the new settings UI too.

yo2ldk, 2016-07-30:
nice add stereo option, Thank you !
 but, if we are in any other mode, am usb,lsb, if I push "F" taste, remain in selected mode not changing to FM.
Also if you can made user colour display, that will be great, and fill spectrum ...etc.  :)
 all the best !

closed 2016-08-16 by cjcliffe