cjcliffe/CubicSDR #413: BladeRF not working with CubicSDR 0.2.0 Beta RC4 on OSX 10.11.3

spaceneedle, 2016-07-26:
I have been trying to get CubicSDR to work with BladeRF on Mac OS X 10.11.3. The device is plugged into a USB 3 port on a new Mac Mini. Unfortunately, when I select the local device in the SoapySDR dialog and click start, nothing happens and the window remains open. I tried all of the builds back to when SoapySDR was first included, and I was unable to get any of them to function.

I did see the following messages when I ran it in terminal: 

  backend = libusb
  device = 0x14:0x17
  driver = bladerf
  instance = 0
  serial = 61400ddbb2f1814442332b1909a052f4
Make device 0
[INFO] bladerf_open_with_devinfo()
[INFO] bladerf_get_serial() = 61400ddbb2f1814442332b1909a052f4
[ERROR] bladerf_get_timestamp() returned -1 - An unexpected failure occurred
Error making device: getHardwareTime() -1 - An unexpected failure occurred

I know the BladeRF works as I have some other unrelated software running on a Linux machine that works fine with it.

guruofquality, 2016-07-26:
You might want to double check ``bladeRF-cli --version`` between the different systems. I think the software and matching fpga+fw images have been under development. It could be a software/fw mismatch. Just a guess.

spaceneedle, 2016-07-26:
I do not have any blade RF software installed on the mac. Blade RF support is self contained within CubicSDR, correct? Does anyone know what fpga+fw image is known to work with CubicSDR on OS X?

cjcliffe, 2016-07-26:
@spaceneedle do any of the previous versions work with your bladeRF? I did recently update the internal bladeRF host library recently perhaps there was an FPGA update that goes with it; if that's the case an earlier build like rc1 or before might work.

spaceneedle, 2016-07-26:
@cjcliffe I actually tried every single OS X build back to when SoapySDR was first included, and none of them were able to talk to the Blade RF. This blade is about a year old and I run a copy of YateBTS on an Ubuntu machine occasionally, so it could just have old firmware. Do you think I could try to upgrade it to the most recent release?

cjcliffe, 2016-07-26:
@spaceneedle I don't have a bladeRF handy and am unsure about the update details without research but I believe @guruofquality has one and might know where to start.

guruofquality, 2016-07-27:
@spaceneedle I don't have an OSX box to test on specifically, but I'm almost positive its a fw mismatch. I would double check ``bladeRF-cli --version`` which should have come with the libbladerf install. Its probably either recent master branch or a release tag. The folks over at #bladerf on freenode should be able to help figure out which images to burn on the device based on the library software version.

closed 2016-09-02 by cjcliffe