cjcliffe/CubicSDR #415: Can't connect to remote USRP E310 with SoapyRemote.

peterg10, 2016-07-27:
I have a USRP e310 with SoapyRemote, SoapySDR and SoapyUHD built and installed locally on the USRP. 

On the client (a Macbook) with SoapyUtil, SoapyUHD, etc installed I can query the remote server on the CLI:
```$ SoapySDRUtil --find="remote=10.80.2.122:3000"
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Mac OS; Clang version 7.3.0 (clang-703.0.31); Boost_105900; UHD_003.009.004-MacPorts-Release

Found device 0
  driver = remote
  name = 
  node = /dev/axi_fpga
  product = 30674
  remote = tcp://10.80.2.122:3000
  remote:driver = uhd
  remote:type = e3x0
  serial = Fxxxxx
```
However when I start CubicSDR the E310 isn't found. I've tried adding it manually with `10.80.2.122:3000`,`remote=10.80.2.122:3000` and `tcp://10.80.2.122:3000` (the last is URL the server daemon provides) no luck.

What am I doing wrong?

cjcliffe, 2016-07-28:
@peterg10 try building SoapySDR and SoapyRemote using the 'maint' branches i.e. ```git checkout maint``` or ```git clone http://github.com/pothosware/SoapySDR -b maint```

CubicSDR releases are built on the SoapySDR maint branches so that they can maintain compatibility as long as possible before being affected by major API/ABI changes; it's possible you just have a newer and possibly incompatible version on the USRP.

guruofquality, 2016-07-28:
Just for the info: Currently maint/master are both on the same ABI. I haven't yet added any breaking features to master. And the remote protocol has not changed at all. So i would expect different versions to be able to communicate.

I think that if ``SoapySDRUtil --find="remote=10.80.2.122:3000"`` was working on the local machine, then that should mean that either CubicSDR didnt load the remote support module or somehow the args were not specified correctly within Cubic?

I think the args should be "driver=remote,remote=10.80.2.122:3000" to make sure its getting the SoapyRemote plugin and not some other support module. And is there a way to check which modules cubic loaded?

cjcliffe, 2016-07-28:
@guruofquality @peterg10 SoapySDR module statuses should be listed in the terminal output in OSX / Linux when run from command line and stdout in windows which might require a > redirect to .txt file ```CubicSDR.exe > errorlog.txt```

closed 2026-11-14 by cjcliffe