cjcliffe/CubicSDR #419: install fail

yo2ldk, 2016-07-30:
Hi,

   on wxWidgets I get an error 
 wxWidgets-3.1.0$ ./autogen.sh    say autoconf: not found

Please some help

 thank you !

cjcliffe, 2016-08-01:
@yo2ldk make sure you've installed the necessary build tools, on ubuntu based systems it's the following:

```$ sudo apt-get install git build-essential automake cmake
$ sudo apt-get install libpulse-dev libgtk-3-dev
$ sudo apt-get install freeglut3 freeglut3-dev```

From the instructions at: 
https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux

yo2ldk, 2016-08-04:
Thank you for reply Charles,

I do all this, I cannot install wxWidgets,
 files from script I think noy exist anymore in wx folder

closed 2016-09-02 by cjcliffe