cjcliffe/CubicSDR #423: Master branch: new squelch policy not working likely because IQ level vs. Audio level

vsonnier, 2016-08-13:
@cjcliffe Hi ! I've tested the new squelch management, but it doesn't work well for me.
A video to demonstrate the problem:
https://www.youtube.com/watch?v=ZqkiRoE0NNw

This is NFM listening, quite powerful against the noise floor (>-15dB with a floor at -85dB).
The moment the transmission stops, the very loud "audio noise" > normal "voice" actually prevents the squelch to work.

I think the trouble lies in the fact that we are comparing apples and oranges when we compare audio dBs and IQ dBs to select the squelch level in DemodulatorThread.cpp (161):
````cpp
currentSignalLevel = iqSignalLevel>audioSignalLevel?iqSignalLevel:audioSignalLevel;
````    
I can see that you introduce the change on purpose for likely good reason, but I don't think those quantities can be compared. In this use case above, an "audio noise" has much more apparent "power" than the matching IQ "power", nullifying the squelch..

So I tried to only use the IQData like before the squelch changes, to see what it does:
``````cpp
//TEST:
currentSignalLevel = iqSignalLevel;
``````
You see the result in the following video:
https://www.youtube.com/watch?v=q-yb0D2modw

Funny thing, the squelch is perfect there, and much better than before your squelch improvement branch anyway.

All in all, I don't see how IQ power and Audio power can really be compared to each other, their nature is different...

cjcliffe, 2016-08-13:
It's been somewhat balanced here but it was a struggle to get it to work well with POCSAG / DMR here too; I have a feeling if I change my gains it won't work as well.

I'll change it for now so that FM based uses the I/Q signal and AM based uses the audio signal -- better AM, LSB and USB responses were the primary reason for changing.

cjcliffe, 2016-08-13:
@vsonnier should be patched up now.

vsonnier, 2016-08-13:
@cjcliffe It works rater well for AM, tested 131-138Mhz, 8khz large plane communications: Signal a drowned into noise, and do not occur very often. Still, the squelch managed to mute most of the noise without masking the real transmissions... it is certainly better than before. 
I've added some small simplifications to it with #425.

ve1jot, 2016-08-13:
I've found that the issue, at least from my viewpoint, is that the squelch control was too coarse..especially at lower gain levels...ie not very sensitive RF signal-wise....the first "amount" or gradient of squelch, or first "click" of rotary mouse wheel when in-focus engages way too much squelch, and weaker signals are muted due to this...I wonder if squelch adjustment was a finer gradient, with an exponential, not linear increase as more squelch is added would be an easier solution?

cjcliffe, 2016-08-13:
@ve1jot in the latest commits to master the signal floor and ceiling will try to adjust dynamically to signal conditions instead of having a static range; for me this takes care of most of the "floored" signals that were nearly impossible to see /  set properly.

ve1jot, 2016-08-14:
@cjcliffe Awsome solution...pulled latest and squelch now works very well, at least for me in my main NBFM and FM modes...will try ssb later with downconverter..ty!

cjcliffe, 2016-08-15:
@ve1jot @vsonnier everything is working well here too -- I've spent an hour or so each night trying it out with AM/LSB/USB ham bands and Pagers, DMR, local busses, etc. and all seems better than before.

Going to close this one up for now; thanks for the help!

closed 2016-08-15 by cjcliffe