cjcliffe/CubicSDR #424: Loading saved sesion doesnt work

Torrio888, 2016-08-13:
I saved session to desktop on Linux Mint but when I try to open it it doesn't list the file.

cjcliffe, 2016-08-13:
@Torrio888 did it save without an .xml extension?  I believe the open dialog is looking for .XML files only unless there's an 'All' type option below.

Torrio888, 2016-08-13:
@cjcliffe  yes it saved without .xml extension even tough  save dialog says that it will save as .xml when mannually typing .xml to the file name when saving than it works correctly.

cjcliffe, 2016-08-13:
@Torrio888 interesting; Glad that's all it is -- I can probably add a check to make sure the extension was added properly.

cjcliffe, 2016-08-16:
@Torrio888 in master branch and next release ".xml" extension is now added automatically if it somehow wasn't part of the filename.

Cheers!

closed 2016-08-16 by cjcliffe