cjcliffe/CubicSDR #429: Strange tuning behavior in CubicSDR caused by other tuning software

BuddyPage, 2016-08-14:
Hello all,

This is my first time here and hope this is something you've seen or know how to fix.
Ran into a strange error in the last few days. CubicSDR has been working good on both Mac OS X El Capitan and Windows 10 for me. In recent days I have installed several multimedia tools to evaluate for use with a Diamond 750 USB TV tuner dongle I just ordered (but haven't received yet). 

NextPVR was not a problem, nor did any of the others seem to be, until after I installed Media Portal (http://www.team-mediaportal.com), tuning with the mouse in CubicSDR was broken. The FFT and Waterfall positions of the filter are no longer in sync and when I click on a signal, it always jumps and homes to another location, but never where I click.

I also just noticed that while the filter position in the waterfall  flows smoothly as it should, the FFT position jumps from one precise primary frequency to the next. For example, when sliding the mouse through the FM frequencies in the waterfall, the filter in the FFT jumps directly from 98.000000 to 97.000000 to 96.0000000 and so on. It will not home to anything selected in the waterfall, and always homes and stops on a precise primary frequency, and nothing in between. Directly changing the frequency still works, but with quirks.

I removed all of the TV media software and reinstalled CubSDR multiple times and the problem still persists. It appears that the tuning in CubSDR was broken by one of the software packages designed to use TV Tuner dongles and boards. I say appears because I have been hacked in the past and have to keep that option in the mix as I troubleshoot.

SDR-Radio, nor SDR Sharp have the same problem, though SDR-Radio displayed other strange behaviors with windowing and freezing that went away after resetting defaults and deleting Media Portal and other similar software.

I experience the same problems with CubicSDR under Windows 10 whether using the HackRF One or SDRPlay units. I am assuming there is a system level TV tuner driver or something that I have inadvertently modified. I'll have time to delve into this further later this week, but thought you might have an idea or at least find this information useful as you proceed.

Thanks

cjcliffe, 2016-08-14:
@BuddyPage Sorry if it's confusing but it's actually a feature that you may have enabled without realizing; you can find more information about the "Frequency Snap" feature here:

http://cubicsdr.readthedocs.io/en/latest/application-window.html#tuning-bar

To reset it to 1Hz increments (as it was when you first ran it) right-click the right-most frequency digit (ones column) or the one currently outlined in red.

Frequency snap is useful for making sure you can click / drag exact frequency intervals; the nudge keys '[' and ']' also make use of it.

BuddyPage, 2016-08-15:
Glad to see this. This feature crossed my mind since I use it in SDRSharp, and I even looked for a way to adjust it. Now I know. I will still have to validate that it works later tonight since the Media Portal 2 install caused problems elsewhere too. Perhaps the timing was merely coincidental.

 Glad to see that this problem might not be linked to the TV Tuner software, since I planned to run it on the same computer with CubicSDR and other SDR software for an upcoming project. 

Thanks for the quick response and glad it is a feature and not a bug-; )

 Keep up the good work.

closed 2016-08-15 by cjcliffe