cjcliffe/CubicSDR #433: Beachball on Macbook Pro on device selection

henrybehmann, 2016-08-26:
I have the latest install (0.2.0 r6) and when selecting the generic RTL dongle the dreaded beachball appears and locks up the app. I have tried reinstalling the app but no luck. This was an intermittent problem in the past but it would eventually work after multiple starts. Any suggestions? Otherwise everything works well. However maintaining a frequency file eludes me. There is no way I can see to list and select the demodulators. I love the program when it works.

Cheers, Henry VE3MTL

henrybehmann, 2016-08-27:
I bite my tongue. After rebooting the Macbook CubicSDR works just fine. So it has to be a Mac problem. Sorry for the premature complaint. All is well.

Cheers, Henry

cjcliffe, 2016-08-28:
No problem, glad it was as simple as that!

closed 2016-08-28 by cjcliffe