cjcliffe/CubicSDR #434: Help! Beta RC6.

lon55, 2016-08-29:
I love this update. I works very smoothly and I have exercised every function for the last two days without a crash running windows 8.

However, I can't figure out how to pipe audio to other apps via VAC. There doesn't seem to be a function in the audio pull-down box, unless I am just missing it.

cjcliffe, 2016-08-29:
@lon55 the "audio output" per-modem is now in the modem properties box beside the modulation buttons -- if the outputs aren't showing in that drop-down list when you open it there might be a problem with the audio device enumeration -- can you confirm the VACs are showing in other applications?

Try running from the command line like so:
```
cd "C:\Program Files\CubicSDR\"
CubicSDR.exe > cubic_output.txt
```

And then quit right away and paste the contents of cubic_output.txt in this issue and we should see the audio device init list in there.

lon55, 2016-08-30:
It was simpler than this, I believe:

[image]

lon55, 2016-08-30:
The tab for Audio Out is the Modem Properties tab you were referring to, I'm guessing.  Everything is working fine.  Thanks much

closed 2016-08-30 by lon55

cjcliffe, 2016-08-30:
yup that's it -- glad it wasn't something more serious; I'll have some more details in the 0.2.0 manual release :)