cjcliffe/CubicSDR #435: Crash when loading saved session

khaytsus, 2016-09-01:
I have created a session with 14 demodulators, ran this way for about an hour or so with no problems.  I made sure to save the session, exited Cubic, set up a shortcut in my launcher to fire up Cubic and re-loaded the session after selecting the SDR and it crashes every time I do this.  So I cannot reload the session I've created.  I've also recreated this with a fresh launch, created one single demodulator, saved session, and attempted to re-load it.  So it appears that this crash is consistent any time I attempt to re-load a session.  The sessions are being saved in my homedir.

I'm running Fedora 24, 64 bit, XFCE.  Using an RTL-SDR, 2mhz bandwidth, PulseAudio output.  Running v0.2.0 AppImage downloaded from the website.  Not sure what other information might be of use.  I can create strace output etc if that's useful.

Here's some output, let me know if you need more?  Thanks!

  setsockopt(IPV6_ADD_MEMBERSHIP) [19: No such device]
Available vertical sync SwapInterval functions:
        glxSwapIntervalEXT: Yes
        DRI2SwapInterval: No
        glxSwapIntervalMESA: No
        glxSwapIntervalSGI: Yes
Using glxSwapIntervalEXT.

Loaded font 'Bitstream Vera Sans Mono' from '/tmp/.mount_8Mgqfr/usr/bin/fonts/vera_sans_mono12_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/tmp/.mount_8Mgqfr/usr/bin/fonts/vera_sans_mono16_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/tmp/.mount_8Mgqfr/usr/bin/fonts/vera_sans_mono18_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/tmp/.mount_8Mgqfr/usr/bin/fonts/vera_sans_mono24_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/tmp/.mount_8Mgqfr/usr/bin/fonts/vera_sans_mono27_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/tmp/.mount_8Mgqfr/usr/bin/fonts/vera_sans_mono32_0.png', parsed 255 characters.
Found Rafael Micro R820T tuner
Loaded font 'Bitstream Vera Sans Mono' from '/tmp/.mount_8Mgqfr/usr/bin/fonts/vera_sans_mono36_0.png', parsed 255 characters.
  default_input = True
  default_output = True
  device_id = 0
  driver = audio
  label = PulseAudio
Make device 0
  device_id=0
  origin=https://github.com/pothosware/SoapyAudio

  available = Yes
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 00000001
  manufacturer = Realtek
  product = RTL2838UHIDIR
  rtl = 0
  serial = 00000001
  tuner = Rafael Micro R820T
Make device 1
Found Rafael Micro R820T tuner
  origin=https://github.com/pothosware/SoapyRTLSDR
  rtl=0

Reporting enumeration complete.
SDR enumerator done.
Found Rafael Micro R820T tuner
SDR thread starting.
device init()
[INFO] Using format CF32.
Stream MTU: 8192
[R82XX] PLL not locked!
Disabled direct sampling mode
starting readLoop()

[Here I load the session]

Terminating SDR thread..
readLoop() ended.
device deinit()
SDR thread done.
Terminating SDR post-processing thread..
Terminating All Demodulators..
Terminating Visual Processor threads..
*** Caught unhandled unknown exception; terminating

cjcliffe, 2016-09-02:
@khaytsus thanks -- I'll try a minimal build here on fedora 24 and see what the debugger spits out.

khaytsus, 2016-09-02:
@cjcliffe If you can't repro it and you can package it or instruct me how to use it, I'll be happy to run any debug stuff here as well.

vsonnier, 2016-09-03:
@cjcliffe @khaytsus If you are running v0.2.0, I've likely found the crash cause, and fixed it by #437

cjcliffe, 2016-09-03:
@khaytsus 0.2.0 Linux and OSX builds have now been updated with @vsonnier 's fix -- Windows builds will be up-to-date shortly and I'll add a note to the releases page.

khaytsus, 2016-09-03:
Awesome, loaded up a few sessions, all is good now!

closed 2016-09-03 by khaytsus