cjcliffe/CubicSDR #441: Main and Modem waterfall out of sync?

Toontje, 2016-09-04:
Check this video: https://files.fm/u/v5kttsun
Why do i have a spike in the main waterfall and i see nothing in the modem waterfall?

cjcliffe, 2016-09-05:
If you're zoomed in it's probably a strong alias from nearby ( see #338 ); tap the C key in a few places on the waterfall -- if it moves around it's an alias.

Toontje, 2016-10-02:
The C key doesn't let it move around. The C key actually syncs the modem waterfall with the main waterfall, which i thought was done automatically.
So now that we know this, how about a Notch filter to get these spikes out?

cjcliffe, 2016-11-20:
C just sets the center frequency which may move the aliases out of the way; hovering the modem and pressing 'C' will always center the modem to the modem frequency.

Notch filters will be covered in #214 with some sort of visual UI control

closed 2016-11-20 by cjcliffe