cjcliffe/CubicSDR #442: CubicSDR plug-ins for signal decoding - PULL REQUEST

adamkk, 2016-09-05:
Hi,

Is there an option to add plug-ins to this software (like plugins to Firefox browser) ? This would enable a modular eco-system like gnu radio.

I am interested in DAB+ signal reception, yet it is troublesome under Win7 (ZADIG drive change every time DAB+ or HDSR is used...).

Ideal option is to add one plug-in to CubicSDR and start decoding it (with a laptop speaker or TCP-IP output).

In case someone would be interested in DVB-T or APRS, then another plugin could start this process.

Thanks :-)

TheButterZone, 2019-02-08:
The kludgy way I do APRS decoding is to feed my Mac Mini/CubicSDR's audio out to my Behringer mixer's RCA/tape in, I punch the button to send that to the RCA/tape out mix which sends it back into my Mac Mini's audio in, then AFSK1200 Decoder 1.0.37. 

Ideally I'd dispense with the mixer & Decoder, and have a CubicSDR plug-in that decodes the packets & relays them to/via APRS-IS without requiring me to have Mensa-level skills like the Mac APRS applications I tried & tried & tried & failed to get to work.

Magalex2x14, 2019-02-08:
@TheButterZone As long as we do not have plug-ins, you should try software audio routing. I personally use Loopback2 app from Rogue Amoeba. Saw free alternatives (Soundflower), but I really like Loopback2.

TheButterZone, 2019-02-08:
Money doesn't grow on trees. I've had the mixer set up like this since I was a teenager, so I can run pro mics & any other audio source in to/out of my computer at once.

Ben1701A, 2021-03-08:
Would love to see an FM-RDS decoder plugin for CubicSDR like gqrx has.