cjcliffe/CubicSDR #443: Linux UHD/USRP Support

erichahn, 2016-09-17:
CubicSDR seriously needs UHD support in linux. The UHD drivers are already supported for linux, when will cubicsdr accept them?

cjcliffe, 2016-09-17:
@erichahn should be bundled with the AppImage and Flatpak 0.2.0 releases -- possibly just throwing an error?

erichahn, 2016-09-18:
The release page https://github.com/cjcliffe/CubicSDR/releases/tag/0.2.0 says that UHD is only supported in Windows and OSX though. After running a USRP with GQRX in linux, closing, and launching Cubic 0.2.0 nothing is discovered. Secondly, adding a device manually has no option for a USRP/UHD device.

erichahn, 2016-09-18:
This probably should be a separate issue, but I also can't get a USRP working with Cubic in Windows 10 either. UHD drivers are installed.

cjcliffe, 2016-09-19:
Ah yes getting ahead of myself -- I forgot there's still some issues with bundling UHD in Linux; but building the module yourself should work with the modpath parameter:

```
--modpath=<str>   Load modules from suppplied path, i.e. '-m ~/SoapyMods/'
```

As for Windows/OSX I can't recall if it's ever been confirmed working with my builds and I don't have hardware to test with -- but I was able to build and bundle the module and it initializes in the log; however that might still be dependent on my build systems somehow.

You may have more luck with CubicSDR builds included with Pothosware for the moment: https://github.com/pothosware/pothos/wiki/Downloads

erichahn, 2016-10-09:
Hopefully this gets figured out soon. I'd love to see people starting to use and develop cubic more. The UHD/USRP users out there would love to build plugins and support this projects.

erichahn, 2016-10-09:
I'm getting it to detect the USRP now, but getting the following errors. 
```

[ERROR] recv packet demuxer unexpected sid 0xfdf41307
O[ERROR] The receive packet handler caught a value exception.
ValueError: bad vrt header or packet fragment
```

Secondly, the sample rate options only go up to 16 MHz, and no text is displayed for the device selection.

[image]

erichahn, 2016-10-09:
UPDATE:
Got it working, but this is just random guessing at settings.
[image]

[image]

Still cannot set sample rate properly.

charlieh0tel, 2017-01-08:
Build CubicSDR, liquid-dsp and SoapySDR from head.

I see the same problem that Eric sees -- no text for the device name under local.

If I select the device and hit start, CubicSDR crashes:

Thread 1 "CubicSDR" received signal SIGSEGV, Segmentation fault.
__memcpy_avx_unaligned () at ../sysdeps/x86_64/multiarch/memcpy-avx-unaligned.S:114
114	../sysdeps/x86_64/multiarch/memcpy-avx-unaligned.S: No such file or directory.
(gdb) where
#0  __memcpy_avx_unaligned () at ../sysdeps/x86_64/multiarch/memcpy-avx-unaligned.S:114
#1  0x00007ffff598f3c2 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_assign(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#2  0x00000000004fd8d4 in SDRDevicesDialog::OnUseSelected(wxMouseEvent&) ()
#3  0x00007ffff6a101fe in wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#4  0x00007ffff6b966c7 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#5  0x00007ffff6b96aae in wxEvtHandler::SearchDynamicEventTable(wxEvent&) () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#6  0x00007ffff6b96b3f in wxEvtHandler::TryHereOnly(wxEvent&) () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#7  0x00007ffff6b96bf3 in wxEvtHandler::ProcessEventLocally(wxEvent&) () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#8  0x00007ffff6b96c55 in wxEvtHandler::ProcessEvent(wxEvent&) () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#9  0x00007ffff6b969c7 in wxEvtHandler::SafelyProcessEvent(wxEvent&) () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#10 0x00007ffff73bf1cb in ?? () from /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#11 0x00007ffff443faec in ?? () from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#12 0x00007ffff3e14fa5 in g_closure_invoke () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#13 0x00007ffff3e26fc1 in ?? () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#14 0x00007ffff3e2f7f9 in g_signal_emit_valist () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#15 0x00007ffff3e3008f in g_signal_emit () from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#16 0x00007ffff45578ec in ?? () from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#17 0x00007ffff443e284 in gtk_propagate_event () from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#18 0x00007ffff443e63b in gtk_main_do_event () from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#19 0x00007ffff40b2c8c in ?? () from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#20 0x00007ffff2e75197 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#21 0x00007ffff2e753f0 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#22 0x00007ffff2e75712 in g_main_loop_run () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#23 0x00007ffff443d697 in gtk_main () from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#24 0x00007ffff7390ed5 in wxGUIEventLoop::DoRun() () from /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0
#25 0x00007ffff6a52343 in wxEventLoopBase::Run() () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#26 0x00007ffff6a17666 in wxAppConsoleBase::MainLoop() () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#27 0x00007ffff6aa3f50 in wxEntry(int&, wchar_t**) () from /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#28 0x0000000000423ca8 in main ()
(gdb) 

I can do a RelWithDebInfo build, if more symbols are needed.

Will investigate on my own, but thought I'd get the bug report in first.

kf7rcs, 2017-11-30:
guys,
 I have a B205 working under gnuradio, but im not how to point cubicsdr to the driver/module.  Do i have to build it for cubic specifically?  
  
erichahn ,  
could you please provide the command line string you used to get your usrp working? 

cjcliffe 
... "building the module  yourself should work with the modpath parameter (available via --help): --modpath=<str>   Load modules from suppplied path, i.e. '-m ~/SoapyMods/'"

could you expand on what this means exactly?

thanks guys,

thanks

erichahn, 2019-02-11:
I've been trying to get this to work again recently and have been having trouble with versions 2.4 and 2.5. Using pybombs now for easier installs/setups but never able to get the uhd modules to work properly even when using the -m option pointing to uhd modules.

Nothing pops up in the device list as it did in the screenshots above.

erichahn, 2019-02-11:
Here are some screenshots of what I've tried. Nothing has made it work or even detect.
[image]

[image]
[image]

erichahn, 2019-02-11:
It's never able to find any "factories" Error with Available factories...No factories found!
Even when it lists libraries that to me might work, but I have no idea.

Dantali0n, 2019-02-15:
The install root of SoapySDR looks wrong being `.////////` not sure if that's an artifact of it being an AppImage or an actual problem.

erichahn, 2019-04-24:
Still hoping that this could be resolved!