cjcliffe/CubicSDR #444: Help with Rig Control

ecj2016, 2016-09-18:
I am setting up CubicSDR for the first time on Win 7 64, using an RTL-SDR to inject the IF signal from my TS-2000, and I want to use rig control and follow.  I have it working with the exception of communicating with my rig.  Other than installing CubicSDR, setting the model, baud rate, and port and then clicking “Enable rig” under rig control is there anything else needed?  Absolutely nothing happens, not even the "PC" indicator on the display of the TS-2000.

Charles, really like this SDR so far.  Thanks for creating it.  I think it is going to be my new favorite spectrum analyzer once the rig control is working.

cjcliffe, 2016-09-22:
@ecj2016 did you have luck fixing the issue?  I haven't had a chance to write the Rig Control docs but will take care of that soon -- I'll keep this open as a reminder to document.

One of the issues here involves re-writing the rig settings so it enumerates the available serial ports for selection instead of taking text input.

ecj2016, 2016-09-23:
No, I haven't had any luck.  I don't think hamlib ever starts.  It would be nice if history would work for SDR-IF.  The TS-2000 uses two IFs depending on which band you are on.

MarkL-W, 2017-01-09:
Is there a guide on persuading Cubic to talk to Rigs ?
I wish to send from Cubic to my Icom R8500. It is connected to the system through the RS232 to usb converter, a standard Prolific 0x2303 chip.

cjcliffe, 2017-01-10:
@ecj2016 @MarkL-W sorry about that -- I've totally forgotten to go back and update the hamlib documentation for the 0.2.0 release.   Will try to get that in soon.

Basically on Windows you'll be using "COM1" "COM2" etc. as a serial port configuration or "/dev/ttyUSB0" or something similar for Linux and OSX -- I need to add proper port enumeration to make this easier.

MarkL-W, 2017-01-10:
Hi Charles,

I forgot to state that I am using Sierra.

I found the "/dev/ttyUSB0" and selected a baud rate of 19200 as this worked
with Mac Memories Manager, but I see no communication between devices.
Perhaps I am missing a setting on the Icom, I will continue to look.

Regaards,
Mark L-W.

ecj2016, 2017-01-10:
When I tried this a while back in Windows 7 and opened a case on it I did use “com2.”  At least in my case it appears that Hamlib does not run because I don’t see the process in task manager.

 

Regards,

 

Ed

W4TBD

colinluthier, 2021-11-26:
Here is what I have sorted out. I found this thread looking for more documentation:

Enable Rig - Check to control the rig. The model, serial rate and control port must be set correctly or you will recieve an error

SDR-IF - ?Set the offset between the SDR frequenct and the radio frequenct for HamitUp like devices????

Control Rig - T\Check this to allow CubicSDR to Control the Rig

Follow Rig -  Check this to allow the rig to control CubicSDR

Floating Center - Allow changes in frequency to change the waterfall display

Track Modem - ???

Model - Set your radio type here. All HamLib radios are supported.

Serial Rate -  This must match the setting in your radio. See the radio manual for the location of this setting.

Control Port - This must be the port the radio is plugged into. ie. dev/ttyUSB0

For me to get a Pan Adapter working I checked Control Rig, Follow Rig, Floating Center and Track Modem. I have it all working but I have to change the mode in both the radio and CubicSDR in order to change the mode.

rszemeti, 2023-01-13:
Struggling to get this to work ... I know my serial connection is good because "other stuff" such as the Kenwoood control program and Minos can talk to the radio, but no luck with Cubic. I have TS2000 selected, the correct baud rate, the right radio and the right serial port, and no other programs trying to use the serial port.  When I check "Enable Rig" nothing happens, no tick appears, no error messages .. nothing.  Ideas?

clouserw, 2023-07-14:
I have a dx3000.  Mine started controlling and tracking the radio after clicking "Track Modem" for what it's worth