cjcliffe/CubicSDR #445: SDRPlay API v1.97.1 disables auto-detect on OS X

Alrescha-git, 2016-09-20:
Just a heads-up, on my system (El Capitan 10.11.6), installing the new SDRPlay API 1.97.1 causes CubicSDR 0.20 to no longer sees the RSP.  Downgrading to 1.95.2 restores functionality.

Alrescha-git, 2016-09-20:
@SDRplay  I'm not in the habit of using your (unsigned) builds, as I generally do not have good luck with them.  That said, I gave your build a try with the new driver/api installed and it says "No SoapySDR modules were found".

The older 0.2.0 build from @cjcliffe does not generate this error, but does not show the RSP with the latest API.  Reverting back to the 1.95.2 driver fixes that problem.

SDRplay, 2016-09-20:
I guess you'll either need to build SoapySDR/SoapySDRPlay like I've done or wait for @cjcliffe to advise. It fundamentally does work though.

SDRplay, 2016-09-20:
If SoapySDRUtil --probe=sdrplay doesn't work then it's possible I've missed something in the installer.

Alrescha-git, 2016-09-20:
@SDRplay I just installed 1.97.1 and your build of CubicSDR on a Macbook that has never seen the software before.  Same result - "No SoapySDR modules were found".

SoapySDRUtil --probe says "driver=null, hardware=null".

SDRplay, 2016-09-20:
Right, what about SoapySDRUtil -info

Can you confirm that it's 0.6 and that the library in /usr/local/lib/SoapySDR/modules has today's date?

Thanks.

Alrescha-git, 2016-09-20:
Info says:

lib version: v0.60-g5b7359a1
api version: v0.6.0
abi version: v0.6-dev
install root: /usr/local
no modules found@
loading modules... done
available factories...null,

SDRplay, 2016-09-20:
no modules found? so nothing in /usr/local/lib/SoapySDR/modules?
that's the problem then I guess. I'll have to check the installer.

Alrescha-git, 2016-09-20:
The modules directory contains one file, libsdrPlaySupport.so, dated 20160920-06:51, 79852 bytes.

SDRplay, 2016-09-20:
so that's weird - it's there but doesn't see it? can you show the listing for /usr/local/lib ?

Alrescha-git, 2016-09-20:
$ ls -l /usr/local/lib
total 976
drwxr-xr-x  3 root  wheel     102 Sep 20 11:00 SoapySDR
lrwxr-xr-x  1 root  wheel      38 Sep 20 11:00 libSoapySDR.0.6-dev.dylib -> /usr/local/lib/libSoapySDR.0.6.0.dylib
-rwxr-xr-x  1 test  staff  200852 Sep 20 06:50 libSoapySDR.0.6.0.dylib
lrwxr-xr-x  1 root  wheel      40 Sep 20 11:00 libSoapySDR.dylib -> /usr/local/lib/libSoapySDR.0.6-dev.dylib
lrw-r--r--  1 root  wheel      36 Sep 20 11:00 libmirsdrapi-rsp.so -> /usr/local/lib/libmirsdrapi-rsp.so.1
lrw-r--r--  1 root  wheel      39 Sep 20 11:00 libmirsdrapi-rsp.so.1 -> /usr/local/lib/libmirsdrapi-rsp.so.1.97
-rwxr-xr-x  1 test  staff  156228 Sep 20 06:21 libmirsdrapi-rsp.so.1.97
-rwxr-xr-x  1 test  staff  111748 Oct 13  2015 libusb-1.0.0.dylib
lrwxr-xr-x  1 root  wheel      33 Sep 20 11:00 libusb-1.0.dylib -> /usr/local/lib/libusb-1.0.0.dylib

SDRplay, 2016-09-20:
OH! The system seems to have changed. Can you rename /usr/local/lib/SoapySDR/modules to /usr/local/lib/SoapySDR/modules0.6-dev - or copy it

SDRplay, 2016-09-20:
Got a feeling that's the wrong libsdrPlaySupport.so you have. I'll have to fix the installer.

Alrescha-git, 2016-09-20:
Linking (or copying) modules to modules0.6-dev and retrying --info results in the error message:

Library not loaded: libmirsdrapi-rsp.so.1.95

SDRplay, 2016-09-20:
yep, building new installer now

SDRplay, 2016-09-20:
ok, 1.97.2 pushed to the website (http://www.sdrplay.com/mac.html) - thanks for your help in getting to the bottom of this.

Alrescha-git, 2016-09-20:
You're very welcome.  I will download and give it a try shortly.

Alrescha-git, 2016-09-20:
I have confirmed that the new driver/API package (1.97.2) works with CubicSDR from the @SDRplay web site.  The 20160831 release from @cjcliffe does not show any devices in the chooser box with the latest API (so, we're back to my original post :)

SDRplay, 2016-09-20:
I just built everything from the latest source available. It's probably the SoapySDR 0.5/0.6 issue - we had a similar problem when SoapySDR went from 0.4 to 0.5 - @cjcliffe rebuilt a new version of CubicSDR and the problem went away. Don't think there's anything more I can do on this. Thanks again.

guruofquality, 2016-09-22:
FYI - I added a print to ``SoapySDRUtil --info`` to show the default module path thats searched:

```
ABI Version: v0.6-dev
Install root: /usr/local
Search path: /usr/local/lib/SoapySDR/modules0.6-dev
```

* The 0.5 releases and maint branch always look in ``$prefix/$libdir/SoapySDR/modules0.5-3``
* The master branch is 0.6 pre-release/development: ``$prefix/$libdir/SoapySDR/modules0.6-dev``
  * In the future when 0.6 ABI series is released: ``$prefix/$libdir/SoapySDR/modules0.6``, which is hopefully a cleaner convention without the extraneous number

The point of this is avoid any ABI clashes between SoapySDR and modules for released versions of the ABI. But to keep it simple, the development tools that get installed with SoapySDR should always get the installation directory correct for the various support modules.

So if thats not happening or there is some other install case that doesnt use the cmake utils that come with SoapySDR, let me know so this can be more streamlined in the future.

Alrescha-git, 2017-06-22:
n.b:  In case it contributes to this problem, I have noticed that the page http://www.sdrplay.com/mac.html is out of date and possibly orphaned.  I have opened ticket #676431 with SDRplay support to let them know.  It may be safer to reference:  http://www.sdrplay.com/downloads/

A.

SDRplay, 2017-06-22:
Link fixed.

No one should be using 1.97 any more - 2.10 is the latest and supports all RSPs

Best regards,

SDRplay Support

vsonnier, 2017-08-25:
By now OSX users have no problems using the current CubicSDR v0.22-master using the latest API v2.10, so I think this issue is no longer relevant.

closed 2017-08-25 by vsonnier