cjcliffe/CubicSDR #446: band-plan hints in the GUI would be nice

richlowe, 2016-09-20:
If the waterfall could be marked up based on a band-plan, that'd be a great time-saver for those of us without an encylopaedic memory, seeing that something "should be" a repeater input, or RTTY, or the like when scanning around.

Dantali0n, 2016-09-21:
I think if this is implemented it would be a good addition to make it user configurable since band-plans can differ based on country and continent.

Toontje, 2016-09-21:
Definitely.

dc1rdb, 2016-09-21:
Maybe this could be combined with https://github.com/cjcliffe/CubicSDR/issues/299

cjcliffe, 2017-09-19:
I think toggling the visibility of ranges might be a reasonable 'general' solution here. Possibly pick a color to go with it for light opacity overlay.