cjcliffe/CubicSDR #448: Locks up on Mac 10.11.6

tsawyer, 2016-10-05:
Running version 0.2.0 locks up my Mac after running for various lengths of time. I wish I could nail the problem down closer but it seems completely random. I can say that one time I got an out of Application Memory message. Restarting CubicSDR does not fix the problem. I have to restart the Mac to get CubicSDR operational again. 

I've not run and prior version of CubicSDR so I can say if the problem is unique to this version. I can run SDR-Console and SDRuno on this same Mac under Parallels with no lockups.

stuzog, 2017-02-06:
On my 10.11.6 iMac 7,1 the precompiled CubicSDR-0.2.0-Darwin.dmg won't boot at all but crashes immediately. 
[170206 Cubic SDR 0.2.0 crash report.txt](https://github.com/cjcliffe/CubicSDR/files/755969/170206.Cubic.SDR.0.2.0.crash.report.txt)

cjcliffe, 2017-02-06:
@stuzog this seems to be a general problem with Core2Duo machines; I think I've got the fixed build but can you give it a try for me?

@tsawyer I think this version may also fix some additional crashes and bugs as it contains @vsonnier 's latest commits:  
[CubicSDR-0.2.2-Darwin.dmg.zip](https://github.com/cjcliffe/CubicSDR/files/756098/CubicSDR-0.2.2-Darwin.dmg.zip)

(note that DMG is zipped to avoid GitHub extension filters)

stuzog, 2017-02-06:
Same behaviour, unfortunately.
[170206 cubicsdr-0.2.2 core2duo crash.txt](https://github.com/cjcliffe/CubicSDR/files/756113/170206.cubicsdr-0.2.2.core2duo.crash.txt)

cjcliffe, 2017-02-06:
@stuzog I have a core2duo here I need to fix up as a test machine which will hopefully give me some more insight.

cjcliffe, 2017-02-06:
@stuzog note that building CubicSDR locally is a known way to get around this crash; it appears to be related to some instructions in liquid-dsp built on my local system that the core2duo machines are not happy with.

stuzog, 2017-02-06:
OK, will try that.

cjcliffe, 2017-02-09:
@stuzog yeah something in liquid-dsp appears incompatible with Core2Do that I hope to track down; I got the old macbook pro finally booted up by restoring an old backup image only to find it's only a CoreDuo limited to 10.6.8.. So unfortunately will have to wait on another system.

stuzog, 2017-04-18:
I spoke too soon: perhaps a system update changed something, but CubicSDR 0.2.0 now crashes immediately on my iMac 7,1. I've been trying to reinstall it but absolutely no luck, even with Hoomebrew. Sigh...

stuzog, 2017-04-18:
Tried again (again) building everything locally. Built, but |cpack| had 
to be run with 'sudo' and threw lots of complaints:

|bash$ cpack
CPack: Create package using DragNDrop
CPack: Install projects
CPack: - Run preinstall target for: CubicSDR
CPack: - Install project: CubicSDR
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: can't open input file: /Users/Stuart/SDR/CubicSDR/build/x64/CubicSDR.app/Contents/MacOS/libfftw3f.3.dylib for writing (Permission denied)
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: can't lseek to offset: 0 in file: /Users/Stuart/SDR/CubicSDR/build/x64/CubicSDR.app/Contents/MacOS/libfftw3f.3.dylib for writing (Bad file descriptor)
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: can't write new headers in file: /Users/Stuart/SDR/CubicSDR/build/x64/CubicSDR.app/Contents/MacOS/libfftw3f.3.dylib (Bad file descriptor)
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: can't close written on input file: /Users/Stuart/SDR/CubicSDR/build/x64/CubicSDR.app/Contents/MacOS/libfftw3f.3.dylib (Bad file descriptor)
CMake Error at /usr/local/Cellar/cmake/3.8.0/share/cmake/Modules/BundleUtilities.cmake:869 (message):
   Command failed:

    'install_name_tool' '-id' '@executable_path/../MacOS/libfftw3f.3.dylib' '/Users/Stuart/SDR/CubicSDR/build/x64/CubicSDR.app/Contents/MacOS/libfftw3f.3.dylib'
Call Stack (most recent call first):
   /usr/local/Cellar/cmake/3.8.0/share/cmake/Modules/BundleUtilities.cmake:943 (fixup_bundle_item)
   /Users/Stuart/SDR/CubicSDR/build/cmake_install.cmake:34 (fixup_bundle)

CPack Error: Error when generating package: CubicSDR
bash$ sudo cpack
Password:
CPack: Create package using DragNDrop
CPack: Install projects
CPack: - Run preinstall target for: CubicSDR
CPack: - Install project: CubicSDR
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: no LC_RPATH load command with path: /Users/Stuart/SDR/wxWidgets/wxWidgets-staticlib/lib found in: /Users/Stuart/SDR/CubicSDR/build/_CPack_Packages/Darwin/DragNDrop/CubicSDR-0.2.2-Darwin/./CubicSDR.app/Contents/MacOS/CubicSDR (for architecture x86_64), required for specified option "-delete_rpath /Users/Stuart/SDR/wxWidgets/wxWidgets-staticlib/lib"
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: no LC_RPATH load command with path: /opt/local/lib found in: /Users/Stuart/SDR/CubicSDR/build/_CPack_Packages/Darwin/DragNDrop/CubicSDR-0.2.2-Darwin/./CubicSDR.app/Contents/MacOS/CubicSDR (for architecture x86_64), required for specified option "-delete_rpath /opt/local/lib"
CPack: Create package
CPack: - package: /Users/Stuart/SDR/CubicSDR/build/CubicSDR-0.2.2-Darwin.dmg generated.
CPack: Create package using STGZ
CPack: Install projects
CPack: - Run preinstall target for: CubicSDR
CPack: - Install project: CubicSDR
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: no LC_RPATH load command with path: /Users/Stuart/SDR/wxWidgets/wxWidgets-staticlib/lib found in: /Users/Stuart/SDR/CubicSDR/build/_CPack_Packages/Darwin/STGZ/CubicSDR-0.2.2-Darwin/./CubicSDR.app/Contents/MacOS/CubicSDR (for architecture x86_64), required for specified option "-delete_rpath /Users/Stuart/SDR/wxWidgets/wxWidgets-staticlib/lib"
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: no LC_RPATH load command with path: /opt/local/lib found in: /Users/Stuart/SDR/CubicSDR/build/_CPack_Packages/Darwin/STGZ/CubicSDR-0.2.2-Darwin/./CubicSDR.app/Contents/MacOS/CubicSDR (for architecture x86_64), required for specified option "-delete_rpath /opt/local/lib"
CPack: Create package
CPack: - package: /Users/Stuart/SDR/CubicSDR/build/CubicSDR-0.2.2-Darwin.sh generated.
CPack: Create package using TGZ
CPack: Install projects
CPack: - Run preinstall target for: CubicSDR
CPack: - Install project: CubicSDR
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: no LC_RPATH load command with path: /Users/Stuart/SDR/wxWidgets/wxWidgets-staticlib/lib found in: /Users/Stuart/SDR/CubicSDR/build/_CPack_Packages/Darwin/TGZ/CubicSDR-0.2.2-Darwin/./CubicSDR.app/Contents/MacOS/CubicSDR (for architecture x86_64), required for specified option "-delete_rpath /Users/Stuart/SDR/wxWidgets/wxWidgets-staticlib/lib"
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: no LC_RPATH load command with path: /opt/local/lib found in: /Users/Stuart/SDR/CubicSDR/build/_CPack_Packages/Darwin/TGZ/CubicSDR-0.2.2-Darwin/./CubicSDR.app/Contents/MacOS/CubicSDR (for architecture x86_64), required for specified option "-delete_rpath /opt/local/lib"
CPack: Create package
CPack: - package: /Users/Stuart/SDR/CubicSDR/build/CubicSDR-0.2.2-Darwin.tar.gz generated.
bash$

|

cjcliffe, 2017-04-20:
@stuzog looks like you've written some cmake files with administrator permissions by attempting to package before it was fully built (which would resume or perform the build and subsequently generate intermediate build files as root); you'll need to run something like this to fix them:

```
sudo chown -R Stuart /Users/Stuart/SDR/CubicSDR
```

Before it's going to allow you to update any of the files again.

stuzog, 2017-04-20:
I started again following your build OS X instructions and got to here 
after installing SoapySDR:

bash$ SoapySDRUtil --info
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Lib Version: v0.6.0-g6f52f9fd
API Version: v0.6.0
ABI Version: v0.6-dev
Install root: /usr/local
Search path: /usr/local/lib/SoapySDR/modules0.6-dev
Module found: /usr/local/lib/SoapySDR/modules0.6-dev/librtlsdrSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6-dev/libsdrPlaySupport.so
Loading modules... [ERROR] 
SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules0.6-dev/librtlsdrSupport.so)
dlopen() failed: 
dlopen(/usr/local/lib/SoapySDR/modules0.6-dev/librtlsdrSupport.so, 1): 
Library not loaded: /opt/local/lib/libusb-1.0.0.dylib
Referenced from: /usr/local/opt/librtlsdr/lib/librtlsdr.0.dylib
Reason: image not found
[ERROR] 
SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules0.6-dev/libsdrPlaySupport.so)
dlopen() failed: 
dlopen(/usr/local/lib/SoapySDR/modules0.6-dev/libsdrPlaySupport.so, 1): 
Library not loaded: /usr/local/lib/libusb-1.0.0.dylib
Referenced from: /usr/local/lib/libmirsdrapi-rsp.so.2.10
Reason: image not found
done
Available factories...null,
bash$

cjcliffe, 2017-05-12:
@stuzog I'd try manually clearing out the SoapySDR modules folders and rebuilding SoapySDR and the modules again -- it looks like something is out of sync; the dependencies are missing dependencies..

closed 2026-11-14 by cjcliffe