cjcliffe/CubicSDR #451: Settings buttons on main screen

TonMachielsen, 2016-10-15:
Here is the use case:
I have saved my settings per HAM band i want to watch/listen to. Every time i want to see a different band i need to File, Open Session, Browse for the correct session file and open it.
It would improve my workflow a lot if i could select session files from the main screen. In my case as HAM band selector, but i can imagine that other people have other uses for a preset selector on the main screen.

cjcliffe, 2016-10-16:
@TonMachielsen there's a bookmarks branch in the works which has basic integrated active, bookmark and recents list with drag & drop support and grouping.

I think you'll be able to create groups easily and there will be an option to "open all in group" but I don't see why I couldn't add an option to save the center frequency/view like sessions do as another branch of items in the bookmarks list so you can just double click to jump to them.

cjcliffe, 2016-10-17:
@Toontje Interesting; perhaps some sort of bar at the top of the application where you could pin bookmarks.

Toontje, 2016-10-17:
How about between the modem waterfall and the main waterfall here:

[image]

or, even better, between the frequency display and the main waterfall here:

[image]

cjcliffe, 2016-11-21:
Hmm, perhaps a standard ribbon UI on the top for favourite bookmarks and views and other things; would like to avoid dividing or cluttering the main UI though -- will have to think about that but can at least start with supporting it in the bookmark manager.

Toontje, 2016-11-21:
How about doing it like SDR Touch? You can hide the bookmarks menu where you have the DSB settings now. Like this:
[image]

Toontje, 2016-12-30:
For me this is now covered by the Bookmarks function. As far as i'm concerned you can close this issue.

cjcliffe, 2017-01-02:
Excellent 👍  we can revisit pinning something to a button bar if the UI features become relevant.

closed 2017-01-02 by cjcliffe

Toontje, 2017-01-02:
No need. It’s perfect as it is.