cjcliffe/CubicSDR #452: Bug in rig control?

Toontje, 2016-10-16:
I want to follow my rig in CubicSDR and my settings are the following:
- Enable rig On
- Control rig Off
- Follow rig On
- Floating Center On
- Track Modem On

I want my modem to move when i turn the VFO of the radio. I don't want the radio to go to a different frequency when i change the modem in CubicSDR. I thought i'd achieve that by setting Control rig Off and Follow rig on. However, when i click in the waterfall my radio is set to the same frequency as the waterfall.

Bug?

lon1955, 2016-10-17:
I have a similar problem.  When I Enable Rig, I see the "CAT" function on my rig illuminate for just a second, then it goes off.  The other controls you mention don't have any effect.  I am running a Yaesu FT-900, and have selected the rig, the baud rate and the correct com port.

cjcliffe, 2016-11-03:
I'm thinking rig support is pretty much alpha in general; I need to re-visit the functionality and come up with a proper configuration dialog instead of assuming everything will work with the hamlib defaults.  Some sort of serial emulator or an actual test rig would likely help me get this under control.

Toontje, 2016-11-06:
Let me send you my TS-590. Would that help? ;-)

cjcliffe, 2016-11-20:
@Toontje that would probably do the trick; is that something you're willing to donate? It would be needed for ongoing testing before releases.  It doesn't even have to have a functional RF unit really as long as the CAT interface responds to commands.

Toontje, 2016-11-21:
It's my only radio, so no, that would not be feasible. Sorry.

Dantali0n, 2016-11-21:
Wondering if their exist known good CAT interface emulators or dummies?

Toontje, 2016-11-21:
I will ask on the TS-590 mailinglists.