cjcliffe/CubicSDR #455: Please update OSX build documentation

Toontje, 2016-10-22:
Please add a section on how to build hamlib for CubicSDR.

cjcliffe, 2016-11-03:
Hmm, still working here with my latest hamlib build but I'll do a fresh pull and look into this.

Toontje, 2016-11-13:
```
TonsMacBookPro:build tmachielsen$ cmake ../ -DwxWidgets_CONFIG_EXECUTABLE=/Users/tmachielsen/Dev/wxWidgets-3.1.0/wxWidgets-staticlib/bin/wx-config -DCMAKE_BUILD_TYPE=Release -DBUNDLE_APP=1 -DCPACK_BINARY_DRAGNDROP=1 -DUSE_HAMLIB=1
-- The C compiler identification is AppleClang 8.0.0.8000042
-- The CXX compiler identification is AppleClang 8.0.0.8000042
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- HAMLIB was not found.
CMake Error at CMakeLists.txt:47 (message):
  hamlib development files not found...

-- Configuring incomplete, errors occurred!
See also "/Users/tmachielsen/Dev/CubicSDR/build/CMakeFiles/CMakeOutput.log".
```
BTW, i updated the CubicSDR cmake command to include /Dev/ n the path. I made an error in the change description of the page though. :-(

Toontje, 2016-11-13:
And BTW, same happens when you don't include -DUSE_HAMLIB=1
```
TonsMacBookPro:build tmachielsen$ cmake ../ -DwxWidgets_CONFIG_EXECUTABLE=/Users/tmachielsen/Dev/wxWidgets-3.1.0/wxWidgets-staticlib/bin/wx-config -DCMAKE_BUILD_TYPE=Release -DBUNDLE_APP=1 -DCPACK_BINARY_DRAGNDROP=1
-- HAMLIB was not found.
CMake Error at CMakeLists.txt:47 (message):
  hamlib development files not found...

-- Configuring incomplete, errors occurred!
See also "/Users/tmachielsen/Dev/CubicSDR/build/CMakeFiles/CMakeOutput.log".
```

Toontje, 2016-11-13:
Ok, installed hamlib using MacPorts. Next issue:

```
TonsMacBookPro:build tmachielsen$ cmake ../ -DwxWidgets_CONFIG_EXECUTABLE=/Users/tmachielsen/Dev/wxWidgets-3.1.0/wxWidgets-staticlib/bin/wx-config -DCMAKE_BUILD_TYPE=Release -DBUNDLE_APP=1 -DCPACK_BINARY_DRAGNDROP=1 -DUSE_HAMLIB=1
-- Found hamlib library at: /opt/local/lib/libhamlib.dylib
-- Found hamlib static library at: /opt/local/lib/libhamlib.a
-- Found hamlib static modules: /opt/local/lib/hamlib-alinco.a;/opt/local/lib/hamlib-aor.a;/opt/local/lib/hamlib-ars.a;/opt/local/lib/hamlib-drake.a;/opt/local/lib/hamlib-dummy.a;/opt/local/lib/hamlib-easycomm.a;/opt/local/lib/hamlib-flexradio.a;/opt/local/lib/hamlib-fodtrack.a;/opt/local/lib/hamlib-gs232a.a;/opt/local/lib/hamlib-heathkit.a;/opt/local/lib/hamlib-icom.a;/opt/local/lib/hamlib-jrc.a;/opt/local/lib/hamlib-kachina.a;/opt/local/lib/hamlib-kenwood.a;/opt/local/lib/hamlib-kit.a;/opt/local/lib/hamlib-lowe.a;/opt/local/lib/hamlib-m2.a;/opt/local/lib/hamlib-pcr.a;/opt/local/lib/hamlib-racal.a;/opt/local/lib/hamlib-rft.a;/opt/local/lib/hamlib-rotorez.a;/opt/local/lib/hamlib-rpcrig.a;/opt/local/lib/hamlib-rpcrot.a;/opt/local/lib/hamlib-rs.a;/opt/local/lib/hamlib-sartek.a;/opt/local/lib/hamlib-skanti.a;/opt/local/lib/hamlib-spid.a;/opt/local/lib/hamlib-tapr.a;/opt/local/lib/hamlib-tentec.a;/opt/local/lib/hamlib-tuner.a;/opt/local/lib/hamlib-uniden.a;/opt/local/lib/hamlib-winradio.a;/opt/local/lib/hamlib-wj.a;/opt/local/lib/hamlib-yaesu.a
-- Found hamlib include directory at: /opt/local/include
64 bit compiler detected
-- Found OpenGL: /System/Library/Frameworks/OpenGL.framework  
-- Found LIQUID: /opt/local/lib/libliquid.dylib  
CMake Error at /opt/local/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES
  wxWidgets_INCLUDE_DIRS)
Call Stack (most recent call first):
  /opt/local/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  /opt/local/share/cmake-3.6/Modules/FindwxWidgets.cmake:917 (find_package_handle_standard_args)
  CMakeLists.txt:123 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/tmachielsen/Dev/CubicSDR/build/CMakeFiles/CMakeOutput.log".
```

Toontje, 2016-11-13:
My wx-config is in $HOME/Dev/wxWidgets-3.1.0/ and not in $HOME/Dev/wxWidgets-3.1.0/wxWidgets-staticlib/bin/
Will update the documentation again if the build goes fine.

Toontje, 2016-11-13:
Not perfect yet. cpack command gives me:

```
[ 95%] Building CXX object CMakeFiles/CubicSDR.dir/src/sdr/SoapySDRThread.cpp.o
[ 97%] Building CXX object CMakeFiles/CubicSDR.dir/src/ui/UITestCanvas.cpp.o
[ 98%] Building CXX object CMakeFiles/CubicSDR.dir/src/ui/UITestContext.cpp.o
make[2]: *** No rule to make target `/Users/tmachielsen/Dev/wxWidgets-3.1.0/lib/libwx_osx_cocoau_gl-3.1.a', needed by `x64/CubicSDR.app/Contents/MacOS/CubicSDR'.  Stop.
make[1]: *** [CMakeFiles/CubicSDR.dir/all] Error 2
make: *** [all] Error 2
```

cjcliffe, 2016-11-19:
Hey @Toontje sorry for the delay; there was a typo when the paths were updated for wxWidgets 3.1.0 -- the configure line's install prefix and the CubicSDR build cmake line's wx-config reference didn't line up.

If you tried to move the folders into place after wx was built you might get the error described above since it's already been "installed" to the wrong location.

Re-do the wxWidgets 3.1.0 build using the paths (with your user folder) from the updated wiki instructions and it should work.

cjcliffe, 2016-11-19:
Hey @Toontje it looks like if you used -DUSE_HAMLIB=1 and then called cmake again without -DUSE_HAMLIB=0 it caches the previously enabled state.

So cmake is working correctly it just cached the enable flag when you ran it again; deleting the build folder and running it fresh should solve this.

Toontje, 2016-11-19:
I don't see any wxWidgets-staticlib directory anywhere.

cjcliffe, 2016-11-19:
@Toontje when you build wxWidgets it uses that as it's installation path; it should create the folder during the build.

Toontje, 2016-11-19:
Here is my build script. Basically the same as the parts in your wiki:

##Remove previous install

rm -rf ~/Dev

## wxWidgets

mkdir ~/Dev
cd ~/Dev
**mkdir wxWidgets-build**
**Why is this directory created and not used?**
wget -O wxWidgets-3.1.0.tar.bz2 wget https://github.com/wxWidgets/wxWidgets/releases/download/v3.1.0/wxWidgets-3.1.0.tar.bz2
tar -xvjpf wxWidgets-3.1.0.tar.bz2
cd wxWidgets-3.1.0
./configure --with-opengl --disable-shared --enable-monolithic --with-libjpeg --with-libtiff --with-libpng --with-zlib --with-mac --disable-sdltest --enable-unicode --enable-display --enable-propgrid --disable-webkit --disable-webview --disable-webviewwebkit --with-macosx-version-min=10.9  **--prefix=/Users/tmachielsen/Dev/wxWidgets-3.1.0** CXXFLAGS="-std=c++0x" --with-libiconv=/usr
**Is this the correct prefix path?**
make -j4 && make install

## SoapySDR

cd ~/Dev
git clone https://github.com/pothosware/SoapySDR.git
cd SoapySDR
SoapySDR$ mkdir build
SoapySDR$ cd build
cmake .. -DCMAKE_BUILD_TYPE=Release
make
sudo make install
SoapySDRUtil --info

## SoapyRTLSDR

cd ~/Dev
sudo port install rtl-sdr
git clone https://github.com/pothosware/SoapyRTLSDR.git
cd SoapyRTLSDR
mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Release
make
sudo make install
SoapySDRUtil --probe

##SoapySDRPlay

cd ~/Dev
git clone https://github.com/pothosware/SoapySDRPlay.git
cd SoapySDRPlay
mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Release
make
sudo make install
SoapySDRUtil --probe

##CubicSDR

git clone https://github.com/cjcliffe/CubicSDR.git
cd CubicSDR
mkdir build
cd build
cmake ../ **-DwxWidgets_CONFIG_EXECUTABLE=/Users/tmachielsen/Dev/wxWidgets-3.1.0/bin/wx-config** -DCMAKE_BUILD_TYPE=Release -DBUNDLE_APP=1 -DCPACK_BINARY_DRAGNDROP=1 -DUSE_HAMLIB=1
**Is this the correct config executable path?**
cpack

cjcliffe, 2016-11-19:
@Toontje there was a bug in the instructions I have now fixed -- the way it was you'd be installing wxWidgets into the build folder which wasn't correct; you'll need to change these lines: 

From:
`./configure --with-opengl --disable-shared --enable-monolithic --with-libjpeg --with-libtiff --with-libpng --with-zlib --with-mac --disable-sdltest --enable-unicode --enable-display --enable-propgrid --disable-webkit --disable-webview --disable-webviewwebkit --with-macosx-version-min=10.9 `**`--prefix=/Users/tmachielsen/Dev/wxWidgets-3.1.0`**` CXXFLAGS="-std=c++0x" --with-libiconv=/usr`

To:
`./configure --with-opengl --disable-shared --enable-monolithic --with-libjpeg --with-libtiff --with-libpng --with-zlib --with-mac --disable-sdltest --enable-unicode --enable-display --enable-propgrid --disable-webkit --disable-webview --disable-webviewwebkit --with-macosx-version-min=10.9 `**`--prefix=/Users/tmachielsen/Dev/wxWidgets-staticlib`**` CXXFLAGS="-std=c++0x" --with-libiconv=/usr`

And

From: 

`cmake ../ `**`-DwxWidgets_CONFIG_EXECUTABLE=/Users/tmachielsen/Dev/wxWidgets-3.1.0/bin/wx-config`**` -DCMAKE_BUILD_TYPE=Release -DBUNDLE_APP=1 -DCPACK_BINARY_DRAGNDROP=1 -DUSE_HAMLIB=1`

To:

`cmake ../ `**`-DwxWidgets_CONFIG_EXECUTABLE=/Users/tmachielsen/Dev/wxWidgets-staticlib/bin/wx-config`**` -DCMAKE_BUILD_TYPE=Release -DBUNDLE_APP=1 -DCPACK_BINARY_DRAGNDROP=1 -DUSE_HAMLIB=1`

Pretty sure that was a copy+paste error when updating the instructions for wxWidgets 3.1.0.

Also the 'Why is this directory created and not used?' there was supposed to be a 'cd wxWidgets-build' in there just to separate your wx widgets build into a separate folder.  Wasn't as major of an issue as the incorrect path; but have added that now.

Toontje, 2016-11-19:
I am still stuck with the message mentioned in https://github.com/cjcliffe/CubicSDR/issues/455#issuecomment-260178481

cjcliffe, 2016-11-20:
@Toontje shouldn't be anything looking in /Users/tmachielsen/Dev/wxWidgets-build if you've made the two changes I suggested above and rebuilt; wxWidgets-staticlib is the installation target folder that's created by configuring and building/installing wxWidgets to use it as the install prefix.

Toontje, 2016-11-20:
If i follow your exact instructions i end up with

```
-- Found hamlib include directory at: /opt/local/include
64 bit compiler detected
-- Found OpenGL: /System/Library/Frameworks/OpenGL.framework  
-- Found LIQUID: /opt/local/lib/libliquid.dylib  
CMake Error at /opt/local/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES
  wxWidgets_INCLUDE_DIRS)
Call Stack (most recent call first):
  /opt/local/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  /opt/local/share/cmake-3.6/Modules/FindwxWidgets.cmake:917 (find_package_handle_standard_args)
  CMakeLists.txt:123 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/tmachielsen/Dev/SoapySDRPlay/build/CubicSDR/build/CMakeFiles/CMakeOutput.log".
CPack Error: CPack generator not specified
```

cjcliffe, 2016-11-20:
@Toontje can you post your script to http://gist.github.com/ and I'll see if I can spot anything?

Toontje, 2016-11-21:
https://gist.github.com/Toontje/34d5f61c2e87a96e3f19cc56f8a773fc

cjcliffe, 2016-11-21:
@Toontje still not seeing the issue here; even copied and pasted the relevant parts of your script and substituted my username and it worked.

For interests sake; what's version do you get when you run `cmake --version` ?

I've got:
`
cjmacbook:build ccliffe$ cmake --version
cmake version 3.5.2
`

cjcliffe, 2016-11-21:
@Toontje also have you updated your OSX since your last ports install/rebuild?

Toontje, 2016-11-22:
```
TonsMacBookPro:~ tmachielsen$ cmake --version
cmake version 3.6.3

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

Yes, i upgraded to Sierra and i updated XCode and macports as well.

cjcliffe, 2016-11-26:
@Toontje sorry for the delay; can you give me the output of the following?

```
cjmacbook:~ ccliffe$ sudo port select --summary
Name       Selected  Options
====       ========  =======
cython     none      cython27 none
llvm       none      mp-llvm-3.7 none
...more stuff...
```

Also working on a quick alternate step-by-step using cmake-gui instead of command line which might give us more insight.

cjcliffe, 2016-11-26:
@Toontje here's the info for building using CMake GUI:

Build everything as usual with your script but skip building CubicSDR.

Get https://cmake.org/files/v3.7/cmake-3.7.0-Darwin-x86_64.dmg and do the usual drag+install to apps.

Run the CMake App; and set the source source to the CubicSDR and the build folder to a separate location; and hit the 'configure' button and choose 'Unix Makefiles' and proceed, you should get errors like the following:

[image]

Scroll to find the key 'wxWidgets_CONFIG_EXECUTABLE' and set it where your wx-config binary is (i.e. /Users/ccliffe/Dev/wxWidgets-staticlib/bin/wx-config):

[image]

[image]

Configure again and you shouldn't get any errors:

[image]

Then hit 'Generate' and you should see in the log:

[image]

Then open a Terminal and build CubicSDR:

[image]

Once that's done you should be able to cd to the x64/ directory and run CubicSDR:

[image]

If any of those steps fail the console / terminal errors might give us some insight into the problem

Toontje, 2016-11-27:
Name     Selected  Options
====     ========  =======
python   none      python26-apple python27 python27-apple none
python2  none      python26-apple python27 python27-apple none
python3  none      none
ruby     none      ruby18 none

cjcliffe, 2016-11-30:
@Toontje I can't make sense of that error or why it's looking in the wxWidgets-build folder -- going to have to try upgrade a Mac to Sierra and let you know what I find.

Toontje, 2016-11-30:
WOW! Thanks!

cjcliffe, 2016-12-01:
@Toontje when you get a chance; try updating you script and remove ```--with-macosx-version-min=10.9``` and if you can paste your script to a gist again I'll review.

Toontje, 2016-12-02:
https://gist.github.com/Toontje/382d76a224b59e97b92377ef05b384f0

Same issue without the --with-macosx-version-min parameter.

stuzog, 2016-12-09:
In building wxWidgets on OS X 10.11.6, at `make -j4 && make install` the following warning repeats consistently at each step although the build apparently proceeds (slowly):

```
In file included from ./src/common/stdstream.cpp:16:
In file included from ./include/wx/wxprec.h:42:
In file included from ./include/wx/wx.h:81:
./include/wx/textctrl.h:749:18: warning: 'OnDynamicBind' overrides a member function but is not marked 'override'
      [-Winconsistent-missing-override]
    virtual bool OnDynamicBind(wxDynamicEventTableEntry& entry);
                 ^
./include/wx/event.h:3678:18: note: overridden virtual function is here
    virtual bool OnDynamicBind(wxDynamicEventTableEntry& WXUNUSED(entry))
                 ^
1 warning generated.
```

stuzog, 2016-12-21:
@cjcliffe I don't have MacPorts installed but use Homebrew, so your script threw an error on line 32:

`sudo port install rtl-sdr`

The equivalent Homebrew command would be:

`brew install rtf-sdr` (with modifications as brew updates itself on a new install)

However, another error occurred when I created a new script to continue installing by starting it at your line 42 (rtf-sdr was already previously installed):

```
-- libsdrplay not found.
CMake Error at CMakeLists.txt:16 (message):
  SDRPlay development files not found...

-- Configuring incomplete, errors occurred!
```

I then installed just CubicSDR with that part of your script, which failed with:

```
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: no LC_RPATH load command with path: /Users/stuart/Dev/wxWidgets-staticlib/lib found in: /Users/Stuart/Dev/CubicSDR/build/_CPack_Packages/Darwin/DragNDrop/CubicSDR-0.2.0-Darwin/./CubicSDR.app/Contents/MacOS/CubicSDR (for architecture x86_64), required for specified option "-delete_rpath /Users/stuart/Dev/wxWidgets-staticlib/lib"
```

[CubicSDR build failed.txt](https://github.com/cjcliffe/CubicSDR/files/665493/CubicSDR.build.failed.txt)

cjcliffe, 2016-12-29:
@Toontje still haven't reproduced issue here but haven't upgraded to Sierra yet either -- holding off on updating my only Mac here until I have a chance to upgrade a non-essential one at work to try it out.

vsonnier, 2017-03-03:
@brrr @Toontje  mentioned in #520 the same issue, and workaround on  mac Serria 10.12.3:
"Report "qtkit.h" not found while installing wxWidgets

Resolution :get wxWidgets master git branch rather than 3.1 source code,and use '--disable-qtkit' for configuration"

stuzog, 2017-03-03:
I've been able to install and run CubicSDR on iMac El Capitan 10.11.6 using Homebrew as per instructions at <https://github.com/cjcliffe/CubicSDR/wiki/Build-OSX---Homebrew>.

Gadgeteering, 2022-08-28:
I have managed to build 0.27 using the following instructions
cd  CubicSDR
mkdir build
cd build
brew install liquid-dsp
brew install wxWidgets
brew install soapySDR
cmake ..
make