cjcliffe/CubicSDR #456: [OT] Scripting issue CubicSDR (Applescript)

Toontje, 2016-10-23:
I am almost done setting up my default desktop using Applescript. The only thing missing is the click on the "SDRPlay RSP" option in the "CubicSDR :: SDR Devices window".
I cannot seem to find the objects in the devices list. I checked with the Accessibility Inspector, but also the devices don't show up there. 
What did you do here? Why can i not find the items in the Accessibility Inspector?

cjcliffe, 2016-11-03:
@Toontje I haven't done much with applescript; but if I can I get a sample script that leads up to what you want I'll see what I can do with the UI controls to expose it properly as I assume this affects accessibility in general..

Toontje, 2016-11-06:
Well, i'm not a great Applescript programmer either. This is what i have so far:
`tell application "fldigi-3.23.13"
	activate
end tell

tell application "System Events" to tell process "Fldigi"
	set position of window 1 to {-1920, -871}
	set size of window 1 to {920, 575}
end tell

tell application "CubicSDR"
	activate
end tell

tell application "System Events" to tell process "CubicSDR"
	set position of window 2 to {-1920, -296}
	set size of window 2 to {1635, 500}
end tell

(* tell application "TCP2Serial"
	activate
end tell

tell application "System Events" to tell process "TCP2Serial"
	set position of window 1 to {1920, -871}
	tell window 1
		tell group "Server"
			click button "Start"
		end tell
	end tell
end tell
*)

tell application "RUMlogNG"
	activate
end tell

tell application "System Events" to tell process "RUMlogNG"
	set position of window "Band Map (active)" to {-283, -870}
	set size of window "Band Map (active)" to {280, 1075}
	set frontmost to true
	click menu item "DX-Spots" of menu "Window" of menu bar 1
	set position of window "DX Spots" to {-998, -871}
	set size of window "DX Spots" to {600, 575}
end tell`

Where i get stuck is in the "System Events" tell block for CubicSDR. I want to do something similar to what i do in the last block above. Since CubicSDR is not AppleScript enabled i have to access the screen elements using the system events application. In the last tell block for RUMlogNG you can see i manipulate the windows and menus to simulate menu selections. I want to do something similar in CubicSDR, but i cannot find the correct address for the SDR Devices drop-down list, so i don't know where to "click". The closest i get is to the whole input box for SDR Devices as shown here: https://postimg.org/image/5qv6htnft/
Maybe this is because it's a wx window and not a native OSX object?

Toontje, 2016-11-16:
BTW, this is a low priority. ;-) This is just a nice to have. I've been going through the open requests and you have too much on your plate to spend time on this.

closed 2026-11-14 by cjcliffe