cjcliffe/CubicSDR #458: DRM support

BooBSD, 2016-10-24:
Is it possible to implement DRM (Digital Radio Mondiale) for CubicSDR?

Dantali0n, 2016-10-24:
Just curious is DRM the same thing as DAB / DAB + (Digital Audio Broadcast)?

BooBSD, 2016-10-24:
@Dantali0n 
> Just curious is DRM the same thing as DAB / DAB + (Digital Audio Broadcast)?

No.

smartynov, 2017-04-16:
I wonder if this is possible too. There are several open source software project on DRM, probably the code could be implanted or rewritten.

cjcliffe, 2017-04-20:
@smartynov if there's available C++ code it should be possible to implement as a modem in CubicSDR; might be worth replacing as much DSP portions as possible with liquid-dsp equivalents that are already built-in.

Channel selection, etc. could also be done through the dynamic modem settings interface.

smartynov, 2017-04-22:
Thanks @cjcliffe, I will probably try to dive deeper and find suitable sources (as usual, when I have time, probably in June...)