cjcliffe/CubicSDR #461: What is happening here?

Toontje, 2016-11-19:
Environment: CubicSDR 0.2.0 on OSX 10.12.1 using SDRPlay.
Video file: https://www.easypaste.org/file/TdCtOlWv/Untitled.mov?lang=en

I just uploaded a video file showing a behaviour which i would like to understand. As you can see in the beginning of the video, there is a noise peak around 7.007MHz. This makes that i cannot hear the signals behind it. Now, the strange thing is that i can move this noise peak by moving the cursor to a different frequency and pressing the C(enter) key. You can see this behaviour starting at 22 seconds into the video. I move the cursor and press the C key, then the noise moves to the location where the cursor is.
What causes this behaviour? What can i do about it so this noise peak disappears?

cjcliffe, 2016-11-19:
Looks like the DC spike noise I get when running at low input sample rates with SoapySDRPlay; what's your input rate in that video?

Once you drop to 500Khz input rate or less the visual DC blocking in CubicSDR is disabled; and in some areas of the HF bands I get a similar noise spike to the one you've captured here at that rate.

If I use 1.5M+ input rate the noise is no longer visible here.

Toontje, 2016-11-19:
You are right. I see this when using any sample rate below 500kHz.

cjcliffe, 2016-11-20:
Thought it looked familiar :)  Don't think we can do much about it -- just the way it seems to work at those sample rates.

closed 2016-11-20 by cjcliffe