cjcliffe/CubicSDR #463: Please clarify license

andreasbombe, 2016-11-20:
I get that the license of CubicSDR is GPL-something, but the details are unclear. The source files are missing copyright and license notices which leaves just two things:

* The README says the license is GPL
* The file LICENSE contains a copy of the GPL version 2

It could be assumed that the license is GPL v2, but is it v2 only or v2 and optionally any later version of the GPL? Please clarify your intended licensing.

cjcliffe, 2016-11-20:
@andreasbombe thanks for noticing this; I'll make sure to clarify that it's GPL v2 or later and put the appropriate information in the source headings.

cjcliffe, 2017-01-03:
@andreasbombe I think I've now covered your concerns with the latest commits; let me know if there's any issues outstanding and I'd be happy to clear them up.

closed 2017-01-31 by cjcliffe