cjcliffe/CubicSDR #464: Update on build crash MacPro Tower OSX Fixed.

g0oan, 2016-11-20:
Hi Charles,

You may remember that during the summer I was having a few problems trying to get CubicSDR running on my MacPro tower.

The MacPorts build failed while the Homebrew version compiled and ran just fine.

Having a little time on my hands last weekend I had another go at trying to build the application using the MacPorts instructions.

As expected the build failed and while trying to figure out what was going on I happened to notice something.

It would appear that MacPorts installs the SoapySDR modules into location /usr/local/lib/SoapySDR/ while CubicSDR looks for the modules in directory /opt/local/lib/SoapySDR/

Simply copying the modules folders from /usr/local/lib/SoapySDR/ to /opt/local/lib/SoapySDR/ solved the problem. The MacPorts version now builds as expected and runs just fine.

Hope this helps and thank you once again for all of the time and effort.

Kind regards,

Sean.
G0OAN.

cjcliffe, 2016-11-20:
@g0oan thanks for the update; it sounds possible that you've still got some SoapySDR libs or environment variables in a bash profile from MacPorts kicking around.  It should use the path from the installed library which I believe is located by environment variables or path search.

g0oan, 2016-11-21:
Hi Charles,

I did a little experiment this morning using an external hard drive which has a completely clean install of Mac OS 10.11 on it.

1. Checked that there were no copies of SoapySDR on the drive, there were none.

2. Installed MacPorts and installed SoapySDR using "sudo port install SoapySDR"
MacPorts installed SoapySDR into directory /opt/local/lib/SoapySDR

3. Uninstalled MacPorts, deleted all of the MacPorts directories and checked that there were no copies of SoapySDR on the hard drive, there were none.

4. Installed Homebrew and then installed the CubicSDR HEAD. Then I install the SoapyRTLSDR module using "brew install soapyrtlsdr" That placed an alias file for SoapySDR in the /usr/local/lib/ directory linking to the original file which the install has placed in /usr/local/cellar/SoapyRTLSDR/lib/

As you can see MacPorts installs SoapySDR into the /opt/local/lib path, as it does just about everything else from what I can see, while SoapySDR, either built from the source code or the Homebrew install places SoapySDR into /usr/local/lib/.

I guess what I'm trying to flag here is perhaps the MacPorts version of the CubicSDR build cmake file isn't picking up that it needs to look in a different place for SoapySDR and its modules. For the MacPorts version it needs to look in /opt/local/lib/ as opposed to /usr/local/lib/

Hope this helps and thanks again for such a great program.

Kind regards,

Sean.

stuzog, 2016-12-09:
When I try `brew install soapyrtlsdr` brew cannot find any formula with that name.
OSX 10.11.6

g0oan, 2016-12-11:
Hi Stuart,

I think I may have build SoapyRTLSDR from source, not using MacPorts or HomeBrew.

It's pretty straight forward it you haven't done it before.

Go to [https://github.com/pothosware/SoapyRTLSDR](url) and download the code then follow the instructions to compile it which you'll find at [https://github.com/cjcliffe/CubicSDR/wiki/Build-OSX](url) If you downloaded the code directly from github you can skip the 'clone' command.

After that you might also want to down the source code for SoapyAudio from [https://github.com/pothosware/SoapyAudio](url) which will allow CubicSDR to access the audio inputs. Compile using the same set of commands as SoapyRTLSDR above.

Hope this helps,

Sean.

stuzog, 2016-12-11:
@g0oan Thanks! That set me off on the right track. I followed the link to [Build-OSX](https://github.com/cjcliffe/CubicSDR/wiki/Build-OSX) where there's a (new?) link to [Installing with Homebrew](https://github.com/cjcliffe/CubicSDR/wiki/Build-OSX---Homebrew). I had to symlink a previously-installed `/usr/local/bin/SoapySDRUtil`, after which a reinstall ran perfectly.

g0oan, 2016-12-12:
Good to hear that it worked out for you. 

Sean.

closed 2016-12-26 by cjcliffe