cjcliffe/CubicSDR #466: Can't "hide" CubicSDR app on OS X

x-f, 2016-11-27:
I can't hide the app on OS X (10.10, 10.11) by using ⌘H or from the menubar "CubicSDR -> Hide CubicSDR" — menubar item highlights, but the app does not hide.
Only thing that works is by right-clicking the Dock icon and then clicking "Hide".
This has been so since 0.1.x versions, not introduced in more or less recent changes.

⌘M doesn't minimise the window either, but that's another topic I guess.

cjcliffe, 2016-12-11:
Hmm; that might be something I have to handle like the special quit; thanks for the report!

vsonnier, 2018-02-02:
@cjcliffe  Hi! Since I'm in a cleanup mood (kill that old issues with fire), could you say this issue is still valid ?

cjcliffe, 2018-05-08:
@vsonnier This one is still valid; just tested here -- better to leave it for now so I don't forget about it again :)