cjcliffe/CubicSDR #467: Sometimes no waterfall

Toontje, 2016-12-04:
Are there any specific conditions in which no waterfall is displayed?
[image]

Toontje, 2016-12-24:
[image]

cjcliffe, 2016-12-26:
@Toontje I hit this one the other day myself; I think it was a failure to allocate the GPU memory for the waterfall because of other things running (i.e. Chrome site with heavy GPU usage) and I may not be catching that -- it's been extremely rare here does it happen often for you?

Toontje, 2016-12-27:
What's the difference between often and regularly? It happens regularly, i would say once or twice out of every 5 runs. Is this often? I normally run CubicSDR together with RUMlogNG and fldigi. No browsers open.

cjcliffe, 2016-12-27:
That seems rather often rather than rare (I'd consider 1-2 a week rare with regular use).

If you don't open the other applications does it still happen every few runs?

Toontje, 2016-12-28:
I don't really know. Probably that means that i haven't seen that behaviour when enough resources are available.

starvald, 2017-01-07:
Started getting this a lot in 0.2.1-alpha, running on ubuntu 16.04 on a beefy gaming capable pc.
should have plenty of resources.

```
lee@Mate:~$ inxi
CPU~Hexa core Intel Core i7-5820K (-HT-MCP-) speed/max~2400/3301 MHz Kernel~4.9.0-040900-generic x86_64 Up~13:55 Mem~4138.4/32002.4MB HDD~11508.3GB(46.2% used) Procs~367 Client~Shell inxi~2.2.35  
lee@Mate:~$ inxi -G
Graphics:  Card: NVIDIA GM200 [GeForce GTX 980 Ti]
           Display Server: X.Org 1.18.4 driver: nvidia
           Resolution: 2560x1440@59.95hz, 800x600@85.06hz
           GLX Renderer: GeForce GTX 980 Ti/PCIe/SSE2
           GLX Version: 4.5.0 NVIDIA 375.26

```

Toontje, 2017-01-07:
Yeah, I don't think it is a resource issue. I have plenty of CPU and memory
in my laptop and it seems to happen every version more often.

cjcliffe, 2017-01-11:
Hey @Toontje; @vsonnier has supplied a fix for this which will be in the next build -- Cheers!

closed 2017-01-11 by cjcliffe