cjcliffe/CubicSDR #468: Can't go to full screen

gegtor, 2016-12-08:
I cloased CubicSDR while in full screen mode now when i launch it i can't access top left window buttons

[image]

cjcliffe, 2016-12-11:
Should be able to find the horizontal resize cursor on the right side of the window; click and instead of dragging left-right drag straight up or down and it should activate the window move instead of resize -- from there you should be able to pull the window down into view.

I'll leave this open as a reminder to try and reposition the window and clamp it to visible area on startup.

gegtor, 2016-12-11:
Okay thanks :)

benallard, 2017-01-14:
After numerous search on google how to reset the window size for an application, I found this bug report and your description helped me. Thanks. Can you maybe set a milestone there ?

cjcliffe, 2017-01-16:
@benallard thanks - I do believe that's the proper solution here; I store the 'maximized' flag on exit and will have to add a special case for restoring in windowed mode for MacOS when that's set.

cjcliffe, 2017-01-19:
@gegtor master branch and next release won't save the window position if CubicSDR is full-screen on exit -- this effectively prevents it from re-loading like this.

So CubicSDR will now always start in windowed mode with the same position it last exited in windowed mode.

closed 2017-01-19 by cjcliffe