cjcliffe/CubicSDR #471: What is up with new releases? OS X Sierra support? New API support? SDRPlay 2.0 support?

MiamiC70, 2016-12-17:
What is up with new releases? OS X Sierra support? New API support? SDRPlay 2.0 support?

SDRplay, 2016-12-17:
We're working on SoapySDRPlay on for both RSP1 and RSP2 - hopefully have something to test soon.

cjcliffe, 2016-12-17:
@MiamiC70 should hopefully have an experimental build out with a bookmark manager before the end of the year which will have updated Soapy API support for SDRPlay as well.

MiamiC70, 2016-12-17:
Will that build be supporting the new SDRPlay 2?

cjcliffe, 2016-12-20:
@MiamiC70 I'm unsure as of the compatibility of the current lib with the new API but SoapySDR should at least be up to date to accept the new module when it's ready.

cjcliffe, 2017-01-03:
@MiamiC70 Latest builds at https://github.com/cjcliffe/CubicSDR/releases/tag/0.2.1-alpha1 should work with the RSP1 and RSP2 with SDRPlay API installed on Sierra.

Build guide updates for Sierra will be done soon as well; cheers!

closed 2017-01-03 by cjcliffe

MiamiC70, 2017-01-03:
Excellent,  thank you