cjcliffe/CubicSDR #472: Problem installing CubicSDR on

Venz299, 2016-12-20:


cjcliffe, 2016-12-20:
Ahoy @Venz299 -- there's a few possibilities here and I'm tracking a possibly related issue.

I've forgotten that someone already contributed a converted version of the instructions to the manual site which I'll probably remove and replace with a link to https://github.com/cjcliffe/CubicSDR/wiki/Build-OSX -- if you used the instructions from the manual try these ones instead for wxWidgets and see if the situation improves.

Otherwise can you confirm that `/Users/svenscheffers/wxWidgets/wxWidgets-staticlib/bin/wx-config` exists?  It should have provided  wxWidgets_LIBRARIES
and wxWidgets_INCLUDE_DIRS -- is it possible you missed the `make install` step of building wxWidgets?

It's also possible with recent updates to the build instructions that there's still a problem somewhere -- I was tweaking the paths for wxWidgets which were previously in error; I'll re-read and re-try my own instructions again soon to be sure.

Venz299, 2016-12-21:
Thanks for your quick response. The new instructions are working. I'm now able to run all the steps. without error messages.

But...Unfortunately, when I start the CubicRTL program, a popup appears with the message that the program is unexpectedly stopped. Ignore/report/open again. (but then in Dutch ;)). As I use an RTLSDR dongle, I installed the additional SoapyRTLSDR module, but no joy. Still get the popup message.

Rebooting my Mac and then tried again, still no joy. :'(

cjcliffe, 2016-12-24:
@Venz299 can you remove the SoapyRTLSDR module and try installing SoapyAudio?  That would help see if basic mic input is working and determine if this is an RTL-SDR specific issue.

If SoapyAudio does the same; try removing it too and see what happens with no modules; if it still crashes possibly try re-building SoapySDR itself and then SoapySDR modules and CubicSDR again in that order.

Venz299, 2016-12-26:
Thanks. I tried the first steps without result. Then I removed everything and started again. Before installing CubicSDR I installed the SoapySDR and Soapy RTLSDR. That does the trick. Now It's up and running.

cjcliffe, 2016-12-26:
@Venz299 good to hear; if there was any Soapy stuff that CubicSDR could build against already installed before you built SoapySDR then it would end up incompatible with the final SoapySDR build.

Glad that's all it was and thanks for helping me notice to fix up the documentation links :)

closed 2016-12-26 by cjcliffe