cjcliffe/CubicSDR #473: Alternate solo mode

starvald, 2016-12-23:
enhancement...

If you have a lot of modems set in a busy area it can get quite a jumble.
would it be possible to have alt-solo mode where it locks to the first active signal and releases when it ends.
also with bookmarks coming would it be possible to add priority modems with such a mode.

cjcliffe, 2016-12-26:
@starvald ahoy; I had a similar need and wrote it into CubicSDR as a combination of Solo + Squelch -- not sure if this is exactly what you're looking for; but with the latest release you can:

Enable solo mode and have multiple modems (with squelch level set) that are within view. 

When one of the modems breaks squelch CubicSDR will automatically jump to it and it becomes the active modem until squelch is released for a short period. 

When squelch is released the solo focus will jump to the next active squelch break and so on.

Cheers!

TheButterZone, 2018-04-08:
If only I could get squelch to stop breaking without actual transmissions & making my ears bleed.

https://www.nooelec.com/store/nesdr-smartee-sdr.html