cjcliffe/CubicSDR #474: center frequency not fully adjustable

PaulB47, 2016-12-24:
v 0.2.0beta RC6 (running on the current skywave linux)

I was listening to FM 89.9, with the center freq also set at 89.9 MHz. The frequencies displayed on the window ran from 89.0 to 90.9. If I adjust the center frequency so that 89.9 is on the far right, no problem. But if I move it toward the left, starting with a center frequency of about 90.6, the signal falls apart and the music is trashed. In other words, long before I move 89.9 to the left edge, it is trashed. It happens the same whether I adjust via click and drag or by manipulating the center frequency digits.

PaulB47, 2016-12-24:
I forgot to mention, it was an SDRPlay RSP1.

PaulB47, 2016-12-24:
I tried the same thing with RTL-SDR. Moving to either edge seemed to distort the signal displayed some, but less than with the SDRPlay, and the distortion was never enough to trash the music.

cjcliffe, 2016-12-26:
Hi @PaulB47 -- this sounds a result of the way the polyphase filterbank channelizer works; if you look at issue #295 you'll see a drawing indicating how the channelizer might divide the band:
https://github.com/cjcliffe/CubicSDR/issues/295#issuecomment-188011711

Likely when your FM modem crosses one of the channelizer boundaries it will distort and you'll see the signal distort in the smaller spectrum view on one side or the other; that will happen when you center the modem near one of the channel boundaries.

The size and placement of the channels depends on the SDR device bandwidth and is a maximum of about 500KHz (so usually SampleRate / 500Khz is the ideal channel size).  The outer most channels also overlap each other and a strong signal on the right edge can distort one on the left.

I'm still thinking it might be a good idea to add a "show channelizer bounds" to the display menu to help avoid placing modems directly on the boundaries.  It may be possible to remove this limitation in the future by combining channels on demand.

PaulB47, 2016-12-28:
To the extent I understand issue 295, it looked like a problem of aliasing with the modem waterfall in the presence of zoom. I'm not looking at the waterfalls at all, nor the modem display, nor is there any zoom. I can't detect any aliasing. It's like the left third of the main spectrum is simply losing signal strength - at least that is my impression.

I notice also that when I am listening to a station on the left end of the spectrum, some other radio station comes in. 89.9 happens to be a classical station and it is very clear when this other station is starting to replace it.

I think my sample rate is usually 2MHz. I will have to reboot with skywave to check.

Do you have an SDRPlay RSP1, to verify this? Do you want screenshots to see what I am talking about? I'm not sure I am describing it correctly.

PaulB47, 2016-12-28:
It's verified that my sample rate is 2MHz. Here is 89.9 at the left of the spectrum; that signal is completely gone and is replaced by a distorted version of 91.5 (from the right side of the spectrum), at least from what I can hear:
[image]

PaulB47, 2016-12-28:
Here is 89.9 a bit further right. It's just at this point that I hear distortion starting:
[image]

PaulB47, 2016-12-28:
Here is 89.9 at center:
[image]

PaulB47, 2016-12-28:
Here is 89.9 at the right. It looks like the right sideband (digital?) is rolling off a bit but not affecting the main signal. In fact I can move it farther right so half the main signal is off screen and still the music is clean:
[image]

closed 2016-12-28 by PaulB47

PaulB47, 2016-12-28:
Maybe it is the thing you were mentioning in issue 295. It just struck me as strange I cannot use stations on the left third of the spectrum.

I wonder what those horizontal grey lines are in the spectrum? Some kind of amplitude scale? Yet they move all over...

cjcliffe, 2016-12-29:
@PaulB47 I think it's a combination of hitting the channelizer boundaries, sample rate plus gain appears a bit high;  I'd recommend turning off automatic gain and manually adjusting to get rid of some of the interference first.

Since you said you're at 2Mhz, FM channels that are spaced 400KHz apart (2MHz / 400KHz divides evenly) end up overlapping each other on the edges and appear symmetrical (you can put half a station exactly on either edge); perhaps try a 2.048Mhz sample rate or higher which may interfere less.

Just a hunch; I don't have that many strong FM stations so close together in this area to make a reasonable comparison at the moment.

cjcliffe, 2016-12-29:
@PaulB47 lines on the spectrum are just to give a general sense of whether it's the scale or the signal causing the visuals to change.  When the smaller divisions transition to larger divisions it (should be) a factor of 10 difference.

PaulB47, 2016-12-30:
Sorry, by "2 MHz" I actually meant 2.048MHz (which I assume means 2*1024*1024 Hz). Fooling with manual gain did not seem to do much, even when I turned it down quite a bit. However sample rate did. It seems 2.048 is a worst case. Other rates above and below that still caused rolloff, but it was more symmetrical rather than heavily on the left, and the interference never got so bad that 89.9 was overcome by 91.5. That only happened at sample rate 2.048 MHz. IIRC I tried 750k, 1.056M, 2.048M, 5M and 6M.

BTW notice on that first picture, when I moved 89.9 over to the left, the whole spectrum mushed down, not just the station at 89.9.

One other thing. That issue 295 picture gave the impression that there are multiple boundaries through the spectrum. I never got the impression that it got alternating better and worse as I moved the center freq across. Only that there was a falloff at the two ends.