cjcliffe/CubicSDR #475: Bug in Display>ColorScheme

PaulB47, 2016-12-24:
I went to Display>ColorScheme, which was set to "default". I changed it to "touch", which indeed changed the scheme, apparently to touch style. I then went back to Display>ColorScheme, and instead of seeing the radio button "touch" on, it was still at "default" even though the display is "touch". I clicked "touch" again, then went back to look at Display>ColorScheme again, which now properly said "touch". So it seems there is a delay in the change of state of the Display>ColorScheme readout even if it does modify the color scheme properly.

cjcliffe, 2016-12-26:
Hi @PaulB47 -- it sounds like an easy fix -- what's the operating system you're using?

PaulB47, 2016-12-28:
I'm using the skywave linux bootable cd.

vsonnier, 2017-08-25:
@PaulB47 I tested the latest v2.3 indeed the problem is present. 
On the other hand, the whole UI is not reacting well on mouse inputs and seems to loose events. This is certainly a specific problem to this distribution, because no other Linux users reported such problems, which certainly wouldn't have passed unnoticed. 
I suggesting closing the issue.

vsonnier, 2017-09-04:
Closing it by lack of response an activity.

closed 2017-09-04 by vsonnier

vsonnier, 2017-10-10:
@PaulB47 FYI in the just released Skywave Linux v3.01 CubicSDR is now perfectly fine about the Color Scheme refresh, and all the other UI problems noticed with v2.3 seems gone.