cjcliffe/CubicSDR #477: ARM64 Support?

corq, 2016-12-29:
Building on the PINE64, all goes well with cmake, but fails on make, this is ARM64, (do I need to use special flags?) unsure if just not supported.

 Used RPI API, will attempt again with Linux API download see if it differs. 
```
root@pine64:/opt/SoapySDRPlay/build# cmake ../
-- Build type not specified: defaulting to release.
-- LIBSDRPLAY_INCLUDE_DIRS - /usr/local/include
-- LIBSDRPLAY_LIBRARIES - /usr/local/lib/libmirsdrapi-rsp.so
-- Configuring done
-- Generating done
-- Build files have been written to: /opt/SoapySDRPlay/build
root@pine64:/opt/SoapySDRPlay/build# make
Linking CXX shared module libsdrPlaySupport.so
/usr/local/lib/libmirsdrapi-rsp.so: error adding symbols: File in wrong format
collect2: error: ld returned 1 exit status
CMakeFiles/sdrPlaySupport.dir/build.make:137: recipe for target 'libsdrPlaySupport.so' failed
make[2]: *** [libsdrPlaySupport.so] Error 1
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/sdrPlaySupport.dir/all' failed
make[1]: *** [CMakeFiles/sdrPlaySupport.dir/all] Error 2
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2
root@pine64:/opt/SoapySDRPlay/build# 
```

cjcliffe, 2016-12-29:
@corq I could be mistaken; but I believe that @SDRPlay will have to release a PINE64 compatible lib for it to work?

corq, 2016-12-29:
I realized after posting, that this was likely to be the case. Feel free to
close this and I'll follow up with the sdrplay folks to see if this is in
future plans.

Thanks for responding.

cjcliffe, 2017-01-03:
@corq no problem; cheers!

closed 2017-01-03 by cjcliffe