cjcliffe/CubicSDR #478: Sorting of ranges

Toontje, 2016-12-30:
Awesome work on the bookmarks! Exactly what i was looking for!!

One question: What is the sorting of the ranges based on?
[image]

cjcliffe, 2017-01-02:
@Toontje they're ordered by the center frequency of the view; the visible range and center frequency are separate so it's good to have the center freq in view range (place with 'C') if you're switching sample rates often.

closed 2017-01-31 by cjcliffe

Toontje, 2018-10-13:
Can sorting be made optional?

cjcliffe, 2019-01-04:
@Toontje yup and will probably be the norm once more sorting/management UI is implemented.  

Right now it's not so much 'sorting' it internally that I can turn off as much as it's just the current display criteria which was easiest to implement.